Denvi/Candle #283: Pressing abort button while running job causes random axis movement.

EducatingSavvas, 2019-06-22:
Hiya - I'm using Candle v1.1.7 with a Gradus M1 pro controller with GRBL version 1.1f.20170531. 

I was running a job and went to abort it within Candle, when the spindle plunged into and began cutting the wasteboard. I tried running the same job this time in mid-air and at a random point pressed the abort button again, and a similar thing happened again. Instead of stopping, the machine moves towards the wasteboard. 

Has anyone experienced this before or has any tips how the abort button is working and how it might be doing this? Thanks in advance.

EducatingSavvas, 2019-06-22:
Sorry I just worked it out. I had the gcode set up in MM but the machine information in the settings in inches. Once I changed that back to millimetres the behaviour stopped occurring.

lalalandaudio, 2019-11-04:
Same issue here , let me try this , having all kinds of problems , too deep cuts and this crashing of the spindle into the wasteboard on abort.