Denvi/Candle #286: Compile the Candle 1.17 for 64bits version in Linux Mint x86_64

rmmachado85, 2019-08-14:
Hello,
I imported through Github repo, the Candle 1.17 project to compile the Candle executable in the 64-bit version. 
Initially I had difficulty with the QGLWidget library which I could not find but after installing an additional qt5 package I ended up compiling with success.

Unfortunately, when connecting to the CNC machine, I found that the firmware is GBRL 0.9. I even tried to run a file to cut and it works as expected. But I can't use JOG mode and it's always giving error:
  $ J = G21G91X1Y0Z0F10 <error: Bad number format
$ J = G21G91X0Y-1Z0F10 <error: Bad number format
$ J = G21G91X0Y-1Z0F10 <error: Bad number format

I can not understand. Is it because of the GBRL 0.9 firmware version?

I hope you can help me.

Note: My operating system is Linux Mint 4.15.

ondras12345, 2019-08-18:
> Unfortunately, when connecting to the CNC machine, I found that the firmware is GBRL 0.9. I even tried to run a file to cut and it works as expected. But I can't use JOG mode and it's always giving [...]

Yes, GRBL v1.1 has a new [jogging method](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging).

It was changed in [this](https://github.com/Denvi/Candle/commit/dd63c4fdf9475d4be4939d8f4db27e984ef61b1f) commit (parts of the change may be [here](https://github.com/Denvi/Candle/commit/e327c82f9ba5294b624e60a2a05ddbaa31cdf233)) and a later change is [here](https://github.com/Denvi/Candle/commit/de60c190fc4efdb3bbabd539e0ad89c8bf7bcc31). In short, the used G-codes changed.

#252
> Hello, Candle 1.0.x was the latest versions which works with GRBL v0.9.

rmmachado85, 2019-08-21:
Thank you to reply to me.

I had had a shortcut. I installed Debian 9 distro to solve this situation based on this link: http://naisema.blogspot.com/2017/11/install-candle-on-ubuntu-64bit-machines.html
Because I have a project with the narrow calendar but in future, I will update the GRBL version to test it.

Have a nice day and good job :)

closed 2025-08-30 by github-actions[bot]