Denvi/Candle #290: z axis wrong direction

BOB61, 2019-09-23:
hi,
i bought a 3018 cnc router working with "Candle" Ver. 1.1.7.0.
the port is correcting open and everything works, but when i click the up arrow to move the spindle up, it goes down and viceversa!
is it possible to invert the z axis movement. If so, how?
thank you in advance
Bob

ondras12345, 2019-09-23:
Hi, take a look at GRBL settings (type `$$` to the console to list them).
Which GRBL version does your machine use? [This](https://github.com/gnea/grbl/blob/master/doc/markdown/settings.md) is a list of settings for the latest GRBL.

BOB61, 2019-09-23:
Hi ondras12345,
i tried with G91 G0 Z5, it goes down

ondras12345, 2019-09-23:
Ok, take a look at the $3 setting

BOB61, 2019-09-23:
it is set to 0 ($3=0)

BOB61, 2019-09-23:
I got it! it must be set on $3=7
I thank you so much

BOB61, 2019-09-23:
No, $3=4!
Now is perfect

ondras12345, 2019-09-23:
Ok. Please close this issue.

Icarve, 2020-07-08:
Hi very new here. I have a 1310 machine, the advice above corrected my Z axis issue when I changed to $3=0
However my x axis is still reversed is ther a similar parameter change to correct that?

ondras12345, 2020-07-09:
> Hi very new here. I have a 1310 machine, the advice above corrected my Z axis issue when I changed to $3=0
> However my x axis is still reversed is ther a similar parameter change to correct that?

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Try `$3=1`

Icarve, 2020-07-09:
Thank you very much for your response. I did figure it out once I discovered the chart.

ElGhost1998, 2020-08-01:
Hi I got a cnc 3018 pro and the z axis gos the wrong way when i push up it gose down when I push down it's gose up changed the cables around and still dose it here a video of my problem https://youtu.be/fjBV9mMsYLY
I have not been able to to cut anything because of this .

BlakusSenior, 2020-09-22:
Whilst I am aware of the $ settings I cannot make it actually change the orientation of the axis when I change the value of the setting. Can anyone help on that issue? 

> Hi I got a cnc 3018 pro and the z axis gos the wrong way when i push up it gose down when I push down it's gose up changed the cables around and still dose it here a video of my problem https://youtu.be/fjBV9mMsYLY
> I have not been able to to cut anything because of this .

Check this out and change the $2 setting to =4.  <https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask>

hoshianaaa, 2024-01-13:
[image]

send `$3=1` fix z direction.

closed 2025-09-14 by github-actions[bot]