Denvi/Candle #294: Неактивна консоль

ihtiandr9, 2019-10-14:
Не могу ввести ни одной команды в консоль. она серая( недоступная. также недоступна прокрутка Упр программы , чтоб начать с произвольной строки. ( тоже серая-недоступная)
[image]

ondras12345, 2019-10-14:
Hi, make sure you are connected to the machine.
You can also try renaming the `settings.ini` file in the directory Candle is installed in to something else and see if it solves it. Candle will create a new settings.ini and you'll have to set it up again. The visualiser section will probably be all black after this, just change the colors setting to fix it.

ihtiandr9, 2019-10-15:
Решено . Для этого нужно отключать управление с клавиатуры

Buxxter, 2020-06-22:
Спасибо, дружище! 2 дня мучаюсь с размером карты высот!

closed 2025-09-14 by github-actions[bot]