Denvi/Candle #297: UI improvement request - wide view

gorbln, 2019-10-24:
Сначала по-русски
Уважаемый автор, хотелось бы иметь возможность откреплять панели "перемещение", "консоль" и прочие, и прилеплять их сбоку отдельными окошками, как в autocad или photoshop сделано. Современные мониторы, как правило, широкоформатные, и более логично иметь окно вытянутым в ширину, чем в высоту. При этом хотелось бы иметь в доступе панели с перемещением и консоль, например. А получается, надо прокручивать то на одно, то на другое, что неудобно. Плюс, в линейку прокрутки вообще не попадёшь - она очень тонкая.

Could you please add a possibility to undock the functional panels like "movement"? To get a wide view window without side ruler.
[image]

HataLin, 2019-10-25:
Поставь монитор боком, чувак 🗡

gorbln, 2019-10-25:
> Поставь монитор боком, чувак

Ну, такое, конечно... Особенно ноутбук удобно вертеть.  =D

HataLin, 2019-10-25:
А другого выхода нет - автор забил на всё :)

HataLin, 2019-10-30:
Отлично, спасибо!

iMaks-RS, 2019-10-30:
> 
> 
> Попробуйте следующую версию: [candle_1.2b.zip](https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2b.zip)

Круто! Интерфейс понравился! Был бы признателен, если бы в новой версии появились фишки:

0. После связи с контроллером, проверять параметр $22. Если равен 1 - тогда делать недоступными элементы управления, оставив только "Сброс", "Разблокировка", "Дом". После "хомления" - делать элементы управления активными.
1.  Установка рабочих координат с интерфейса пользователя, а не в ручную. Хотя-бы одну рабочую координату G54 - что-бы после выключения станка, не сбивалось положение рабочего нуля. (Описал здесь: https://github.com/Denvi/Candle/issues/301#issue-513759778)
2. Корректная работа с форком GRBL под STM32 - команды выполняются, УП запускается, но не работает визуализация.. Дальше теста "по воздуху" дело не дошло... Очень привык к удобству Candel ))) (Форк - http://www.cnc-club.ru/forum/viewtopic.php?f=41&t=21393#p477776)
3. Если появится - это будет "отрыв башки": работа с крае-искателем (поиска центра заготовки/отверстий/положение заготовки и корректировка УП).  Есть для линукс в открытом доступе и для Mach3 (http://www.cnc-club.ru/forum/viewtopic.php?f=41&t=15602#p356640)
4. На перспективу - версия для планшетов под Win и для мониторов с тачем.
Спасибо за Ваш труд!

iMaks-RS, 2019-10-31:
Запустил бетту 1.2 на ноутбуке... с разрешением 1366*768... Требуется оптимизация пользовательского интерфейса под это разрешение:
1. Меню настроек - слишком большое по высоте: кнопки "ОК", "Отмена", "По умолчанию" находятся за пределом доступности экрана... даже если сделать плавающую панель задач/меню - кнопки не доступны  - сделать как в 1.1.7
2. Изменить минимальный размер окон "консоль" и "визуализация"
3. Уменьшить размер кнопок - сделать как в 1.1.7
4. А вдруг?! ))  Предложение по изменению расположения элементов:
а) "Консоль" перенести влево вниз - расположить под "Модификации"
б) На место консоли расположить:  "Управление", "Команды", "Шпиндель". Тем самым справа останется "Состояние" и "Перемещение"
По моим прикидкам, такое расположение позволит полностью отобразить все элементы управления для разрешение 1366*768 и исключить необходимость "сворачивать меню".

gorbln, 2019-10-31:
@iMaks-RS 
Вот прямо вот всё написал, что и я хотел! Не прибавить, не отнять. Полностью согласен.
И да, на ноутах сейчас ещё много где 1300х700 (у меня тоже) - интерфейс в целом хорош, но всё, что написал iMaks выше - вот это прям всё оно.
Консоль просится в левый нижний угол, но она слишком широкая - смотрится странненько.

А, да!  Автору - респект! Не бросает пользователей =)

MeJasonT, 2019-10-31:
following thread for UI changes and agree with comments.
Denvi, are you picking up on other UI improvement suggestions as well - still no button for coolant

Denvi, 2019-10-31:
@iMaks-RS 
Сбросьте снимки экрана:
1) как есть на данный момент;
2) как нужно сделать (отредактированные в графическом редакторе).

gorbln, 2019-10-31:
Now
[image]

Suggest
[image]
Width of the columns better to be draggable.
Width of the whole column must be equal.

Хотелось бы, чтобы ширина колонок настраивалась свободно, без привязки к какому-нибудь конкретному окну. Ну и, конечно, ширина всех элементов внутри колонки должна быть одинаковая. И хотелось бы, чтобы можно было открепить любой элемент, а не только группу целиком.

Denvi, 2019-11-01:
1) Минимальные размеры окон "Консоль" и "Визуализатор" сброшены.
2) Окна "Устройства" и "Модификация" имеют изменяемый размер по ширине, но только для размещения вертикальной полосы прокрутки. Получить вменяемый вид этих окон для любой ширины вряд ли получится.
3) Кнопки имеют адаптивный размер.
4) По перетаскиванию элементов окон нужно подумать, нужна ли такая свобода...
5) По экранам с разрешением 1366x768 - решение пунктом выше, подумаю.

Обновленная версия: [candle_1.2.2b.zip](https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2.2b.zip)

HataLin, 2019-11-01:
А можно сделать чтоб джог управлялся курсорными клавишами? на экране кликать не удобно же...

yuranikspb, 2019-11-01:
Добавлю то что сделал для удобства для себя
расчет количества замеров по x и y в зависимости от длинны шага .
[image]

Denvi, 2019-11-01:
> А можно сделать чтоб джог управлялся курсорными клавишами? на экране кликать не удобно же...

Нужно!
В планах сделать настраиваемые быстрые клавиши на все команды/кнопки.

Denvi, 2019-11-01:
> Добавлю то что сделал для удобства для себя
> расчет количества замеров по x и y в зависимости от длинны шага .
> 
> хочется это в основной версии

Первая версия была как раз с настройками шага в мм., возможно имеет смысл ввести переключатель "Количество/шаг".

iMaks-RS, 2019-11-02:
Скрины версии 1.2b

>     1. как есть на данный момент;
[image]
[image]
> 
>     2. как нужно сделать (отредактированные в графическом редакторе).
[image]

В 1.2.2b выглядит уже лучше, но управление на экран не помещается..

[image]

Вот сравнение размера элемента интерфейса для "ноутбучного" разрешения.
[image]

Опционально:
Предложение покажется странным: может имеет смысл  рассмотреть версию для низкого разрешения? Нечто похожее на "стойку" Зафиксировать размер окна визуализации в левом верхнем углу, функциональные кнопки под визуализацией, информация о координатах - справа от окна визуализации, кнопки управления движением - под окном координат. Вот пример недорогой стойки XC609 http://www.xcmcu.com/Uploads/5859eb5b96274.jpg
 Возможно обладателям 14" и 15" такой дизайн придётся по душе, да и б.у. монитор можно будет задействовать.. Есть ведь ещё такая прикольная штука - как ресторанные терминалы, с 14-17 дюймовыми экранами с тачем и встроенным ПК на атоме с парой гигов оперативы. Вполне себе готовое решение.. бу. можно купить, объявлений хватает..

ASHIT-AXAR, 2019-11-03:
in version 1.1.7 there is no cpu load % when the window is minimized 
but in version 1.2.2 it is still using cpu in a constant load even when Candle is minimized.
for instance it is using ~34% cpu for the following file:

[image]

OpenGL 2.1

ASHIT-AXAR, 2019-11-03:
hello @Denvi 
"Restore Origin" only is able to restore if you press "Zero XY" or "Zero Z" buttons.
when you use "Z-Probe" the "Z" value is not saved for restore. (or when Z is set by sending G92Z manualy in console)

dremin1984, 2019-11-06:
И добавить 4 ось. 
Поддерживаю выше сказанное про форк на STM32F103C. На этом мк работает лучше.

dremin1984, 2019-11-06:
> И добавить 4 ось.
> Поддерживаю выше сказанное про форк на STM32F103C. На этом мк работает лучше.

Окна так и оставить, как в версии 1,2,4b. Удобно когда, то, что не нужно в данный момент можно отключить. Визуализатор оставить обязательно. 
зы Просьба сделать сборку для raspbery B3+ сам не могу собрать.  Использую дисплей 7дюймов. Получается как полностью автономный контроллер.

Denvi, 2019-11-06:
В 1.2.5b добавлено окно "Пользователь". Панели можно перемещать внутри/между окнами.

iMaks-RS, 2019-11-06:
Спасибо!  Это даже превзошло все ожидания! Вопрос с разрешением экрана на ноутбуке для меня закрыт.  Появился один момент - можно случайно сдвинуть панель, просьба по возможности добавить блокировку окон, активация например на иконку канцелярской кнопки в заголовке панели.

Denvi, 2019-11-07:
> hello @Denvi , new changes are really nice looking.
> "Restore Origin" only is able to restore "Z" value if you press "Zero Z" button.
> when you use "Z-Probe" the "Z" value is not saved for restore. (or when Z is set by sending G92Z manualy in console)
> what about adding a close button next to the open button (just look at that poor FPS, is there no way to get rid of that other than exit and restart candle app)?

Ok, will fix then. About poor FPS: there is an option "Simplify graphics", should help with it.

XRay3D, 2019-11-14:
https://youtu.be/ltT7e3X1fRI
Убрал избыточные расстановщики и сразу установил min и max ширину с учётом вертикальной прокрутки. Стало адекватнее обрабатывать изменения размера и пропала нижняя прокрутка.

gorbln, 2019-11-29:
@Denvi 
Большое спасибо за Вашу работу! Новый интерфейс великолепен.

Thanks a lot for the improvement of the GUI. New design much more usable!

closed 2019-11-29 by gorbln