Denvi/Candle #299: No-go in Lubuntu 18.04 LTS

mj1911, 2019-10-28:
Working fine on Windows 7 x64.  Not working on Lubuntu 18.04 LTS.  Possibly similar to [issue 79](https://github.com/Denvi/Candle/issues/79), when I try to run Candle, it reports:

```
This application failed to start because it could not find or load the Qt platform plugin "xcb".
Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, xcb.
Reinstalling the application may fix this problem.
Aborted (core dumped)
```

I placed the Candle folder within `~/bin` and symbolic-linked the Candle executable to `~/bin`. Thinking it was legitimately a missing file, I ran `ldd ~/bin/Candle/Candle`:

```
linux-gate.so.1 (0xf7fb6000)
libQt5OpenGL.so.5 => /home/me/bin/Candle/./libs/libQt5OpenGL.so.5 (0xf7f5b000)
libQt5Widgets.so.5 => /home/me/bin/Candle/./libs/libQt5Widgets.so.5 (0xf78e1000)
libQt5Gui.so.5 => /home/me/bin/Candle/./libs/libQt5Gui.so.5 (0xf72ff000)
libQt5SerialPort.so.5 => /home/me/bin/Candle/./libs/libQt5SerialPort.so.5 (0xf72e4000)
libQt5Core.so.5 => /home/me/bin/Candle/./libs/libQt5Core.so.5 (0xf6d9d000)
libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xf6bf0000)
libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xf6aee000)
libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xf6ad0000)
libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xf68f4000)
libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xf68d5000)
libgobject-2.0.so.0 => /usr/lib/i386-linux-gnu/libgobject-2.0.so.0 (0xf6873000)
libglib-2.0.so.0 => /usr/lib/i386-linux-gnu/libglib-2.0.so.0 (0xf6742000)
libX11.so.6 => /usr/lib/i386-linux-gnu/libX11.so.6 (0xf65f8000)
libGL.so.1 => /usr/lib/i386-linux-gnu/libGL.so.1 (0xf6597000)
libicui18n.so.53 => /home/me/bin/Candle/./libs/libicui18n.so.53 (0xf6340000)
libicuuc.so.53 => /home/me/bin/Candle/./libs/libicuuc.so.53 (0xf61c1000)
libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xf61bc000)
libgthread-2.0.so.0 => /usr/lib/i386-linux-gnu/libgthread-2.0.so.0 (0xf61b9000)
librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xf61af000)
/lib/ld-linux.so.2 (0xf7fb8000)
libffi.so.6 => /usr/lib/i386-linux-gnu/libffi.so.6 (0xf61a6000)
libpcre.so.3 => /lib/i386-linux-gnu/libpcre.so.3 (0xf612d000)
libxcb.so.1 => /usr/lib/i386-linux-gnu/libxcb.so.1 (0xf6101000)
libGLX.so.0 => /usr/lib/i386-linux-gnu/libGLX.so.0 (0xf60de000)
libGLdispatch.so.0 => /usr/lib/i386-linux-gnu/libGLdispatch.so.0 (0xf607f000)
libicudata.so.53 => /home/me/bin/Candle/./libs/libicudata.so.53 (0xf4bf6000)
libXau.so.6 => /usr/lib/i386-linux-gnu/libXau.so.6 (0xf4bf2000)
libXdmcp.so.6 => /usr/lib/i386-linux-gnu/libXdmcp.so.6 (0xf4beb000)
libbsd.so.0 => /lib/i386-linux-gnu/libbsd.so.0 (0xf4bd0000)
```

So they all seem to be present, yes?  Is this a version incompatibility then?  None of the [fixes](https://forum.qt.io/topic/67968/missing-qt-platform-plugin-xcb/7) seem to help.  `fixdep.sh` no help. The Qt available in the Lubuntu repository is v5.9.2, but Candle appears to be built with 5.4.2.  I know little about Qt but am skilled in other areas. If it needs to be rebuilt for a newer Qt, can-do, but need a little guidance. Checking with [download.qt.io/official_releases/qt/](http://download.qt.io/official_releases/qt/) only versions 5.9, 5.12, and 5.13 are available. 

Please advise on how to proceed. Thank you.

mj1911, 2019-11-02:
Ok, so I downloaded the source into a different directory, installed cmake 3.15.5, and used `cmake-gui` to try `configure` but it generated a `Could not find a package configuration file provided by "Qt5SerialPort"` message.  Luckily, this was fixable by `sudo apt install libqt5serialport5 libqt5serialport5-dev`. Then it did `configure` and `generate` successfully.

Then I ran a `make` and it built successfully also, so I replaced the original `Candle` executable with the newly-built one, and it seems to be working well.  :)

Hopefully this helps someone else unfamiliar with rebuilding to suit newer Qt versions.

mj1911, 2019-11-11:
Resolved by rebuilding.

closed 2019-11-11 by mj1911

daramos, 2021-09-26:
I also ran into this issue. Seems like the linux pre-compiled release is compiled for x86, not x64.

In my case - I was missing the x86 version of libsm6. Installing it with `sudo apt-get install libsm6:i386` resolved the dependency issue.