Denvi/Candle #301: Изменить алгоритм работы кнопок "ZeroXY", "ZeroZ"

iMaks-RS, 2019-10-29:
Суть вопроса: кнопки "обнулить XY", "обнулить Z" должны обнулять текущие рабочие координаты и сразу записывать текущие машинные как G54.
Цель - после восстановления питания и хомления, по команде с одной из четырёх программируемых пользователем кнопок уходил в координату G54X0Y0Z0

Сейчас это делается в ручную: хомление, подвод в нужную точку, обнуление, запись в ручную в консоле команды G10L2P1XкоординатаYкоородинатаZкоордината. Печаль в том, что нет возможности в самом Candle получить текущие машинные координаты.. даже нет возможности их скопировать - окно не активно. Немного упростил себе жизнь, записав в 2 из 4х доступных кнопок поманды:
Идём в G54 - G54;G0X0Y0;G1Z0F500
Идём в G55, (столик выезжает на меня, шпиндель уходит на вверх и в право) - G55;G0Z0;G0X0Y0

Всё что смог сделать на данный момент.. это с горем по полам установить QTCreatar, QT-5.4.2, minGW. Как-то криво настроить и таки добраться до интерфейса, что бы узнать какие события вызываются по клику и в какие переменные выводятся машинные координаты.. осталось "мелочь".. внести изменения и собрать.. но.. собрать не получается.. и я не уверен что правильно внёс изменения.. т.к. не понятно, являются переменные координат глобальными и можно ли их в текущем времени получить..

вот код изначальный
`void frmMain::on_cmdZeroXY_clicked()
{
    m_settingZeroXY = true;
    sendCommand("G92X0Y0", -1, m_settings->showUICommands());
    sendCommand("$#", -2, m_settings->showUICommands());
}

void frmMain::on_cmdZeroZ_clicked()
{
    m_settingZeroZ = true;
    sendCommand("G92Z0", -1, m_settings->showUICommands());
    sendCommand("$#", -2, m_settings->showUICommands());
}`

а вот моё кривое )) "в лоб" и "на угад"
`    void frmMain::on_cmdZeroXY_clicked()
    {
        m_settingZeroXY = true;
        sendCommand("G92X0Y0;G10L2P1X%txtMPosXY%txtMPosY", -1, m_settings->showUICommands());
        sendCommand("$#", -2, m_settings->showUICommands());
    }

    void frmMain::on_cmdZeroZ_clicked()

    Был бы признателен, если бы кто реализовал этот функционал, собрал проект и выложил в доступ.
    Спасибо.
    {
        m_settingZeroZ = true;
        sendCommand("G92Z0;G10L2P1Z%txtMPosZ", -1, m_settings->showUICommands());
        sendCommand("$#", -2, m_settings->showUICommands());
    }

`

Был бы признателен, за помощь и готовый собранный проект с внесёнными изменениями.
Спасибо.

HataLin, 2019-11-01:
И вообще жаль, что нет кнопки "Go to Home" и что после abort не едет на начальные позиции.

Denvi, 2019-11-03:
Попробуйте версию1.2.3b.
В ней добавлена поддержка вычисляемых выражений внутри команд посредством JavaScript.
Выражения должны быть заключены в фигурные скобки, например:

```
{var a = 10}
````

осуществляет объявление переменной 'a' и присвоение ей значения, равное 10.
В выражениях доступен глобальный объект 'vars', содержащий машинные и рабочие координаты, а также значения всех смещений для систем координат G54-G59, G28, G30, G92, значения координат зонда PRB.

Пример команды для определения центра внутри окружности:

```
G21G91;
{var f = 100}{var r = 10}{var d = 1}G38.2X{r}F{f};
{var q = vars.PRBx}G0X{-d};
G38.2X{-r};
G0X{(q - vars.PRBx) / 2};
G38.2Y{r};
{var q = vars.PRBy}G0Y{-d};
G38.2Y{-r};
G0Y{(q - vars.PRBy) / 2}
```

Для получения рабочих координат можно воспользоваться следующими выражениями:

```
{vars.Wx}
{vars.Wy}
{vars.Wz}
```

Смещения:

```
{vars.G54x}
...
{vars.G54z}
```

Discorez, 2022-08-27:
Подскажите, плиз, где посмотреть весь список доступных переменных объекта vars?

Denvi, 2022-08-27:
Список здесь: https://github.com/Denvi/Candle/blob/Experimental/src/candle/scriptvars.h
Диапазонов координат загруженной УП на данный момент в списке нет.

closed 2025-09-14 by github-actions[bot]