Denvi/Candle #304: Cannot compile version 2.0 against xubuntu 19.10

giasone70, 2019-11-01:
As I try to make, the compiler throws an error and stops:
g++: error: unrecognized command line option ‘-Z7’; did you mean ‘-Z’?
make: *** [Makefile:768: main.o] Error 1

Any suggestion?
Thanks in advance

Denvi, 2019-11-01:
Hi, checkout latest commit. I have removed some compiler options (was for debug purpose, msvc).

giasone70, 2019-11-01:
Thank you very much. It works now.
May you help to get rid of all sources and keep only the files needed to work?

giasone70, 2019-11-01:
Also better it would be if I could convert it in a sort of  already compiled "package".

closed 2019-11-01 by giasone70

ASHIT-AXAR, 2019-11-11:
hello @Denvi 
i couldn't build experimental branch too.
it throws "-Z7" first. i removed the word manually from candle.pro. then the following errors unlocked:

[image]

i replaced all ">>" with "> >" . but the highlighted option still remains. is there a trick ? :thinking:

Denvi, 2019-11-11:
Try replace

```
const char* what() const override
```

with 

```
const char* what() const throw()
```

I haven't mingw now, can't check.

ASHIT-AXAR, 2019-11-11:
that solved the issue. thank you.

shodanx2, 2023-07-14:
Hello,

getting similar issue trying to compile 1.2b on ubuntu 22.04 lts

packages installed 

`sudo apt install qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools libqt5serialport5-dev qtcreator build-essential`

```
cnc@cnc:~/Candle-1.2b/src$ make
g++ -c -pipe -Z -Fdrelease\candle.pdb -O2 -Wall -Wextra -D_REENTRANT -fPIC -DUNIX -DsNan="65536" -DAPP_VERSION=\"1.2b\" -D_USE_MATH_DEFINES -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtSerialPort -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o main.cpp
In file included from main.cpp:18:
frmmain.h:73:17: error: looser exception specification on overriding virtual function ‘virtual const char* CancelException::what() const’
   73 |     const char* what() const override
      |                 ^~~~
In file included from /usr/include/c++/11/new:41,
                 from /usr/include/c++/11/bits/stl_construct.h:59,
                 from /usr/include/c++/11/bits/stl_tempbuf.h:60,
                 from /usr/include/c++/11/bits/stl_algo.h:62,
                 from /usr/include/c++/11/algorithm:62,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:142,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qtguiglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qtwidgetsglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qapplication.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QApplication:1,
                 from main.cpp:4:
/usr/include/c++/11/bits/exception.h:76:5: note: overridden function is ‘virtual const char* std::exception::what() const noexcept’
   76 |     what() const _GLIBCXX_TXN_SAFE_DYN _GLIBCXX_NOTHROW;
      |     ^~~~
main.cpp: In function ‘int main(int, char**)’:
main.cpp:67:8: warning: suggest explicit braces to avoid ambiguous ‘else’ [-Wdangling-else]
   67 |     if (!styleOverrided) foreach (QString str, QStyleFactory::keys()) {
      |        ^
make: *** [Makefile:774: main.o] Error 1
```

For reference line 4 of main.cpp is #include <QDebug>

So I'm not sure where I should try replacing

`const char* what() const override`

to 

`const char* what() const throw()`

mar0x, 2023-07-14:
Hello,
The error message is:
```
frmmain.h:73:17: error: looser exception specification on overriding virtual function ‘virtual const char* CancelException::what() const’
   73 |     const char* what() const override
```
See `frmmain.h` line #73.

Is it possible you are using outdates source files? Current ["Experimental" branch](https://github.com/Denvi/Candle/tree/Experimental) shouldn't have this issue, see
https://github.com/Denvi/Candle/blob/Experimental/src/candle/frmmain.h#L98

shodanx2, 2023-07-15:
Hello,

Indeed I am running the experimetan release 1.2b

https://github.com/Denvi/Candle/releases/tag/v1.2b

I obtained a new copy and it compiled without errors

However there were many warnings, which I include in the following log file

[2023-07-15 candle.pro compile log.md](https://github.com/Denvi/Candle/files/12061820/2023-07-15.candle.pro.compile.log.md)

ubuntu 22.04 build instructions so far would be 

```

sudo apt install qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools libqt5serialport5-dev qtcreator build-essential git
git clone https://github.com/Denvi/Candle.git
cd Candle/src
qmake candle.pro
make
```