Denvi/Candle #307: Abort issue on 3018

lalalandaudio, 2019-11-03:
When I hit abort when using a 3018 it raises the spindle a little bit but then starts moving x/y our of the work area but gradually moves the z down while moving , hitting my material and breaking my bit every time ?

I realize this might be operator error but this is the only place i could find to ask .

Thanks

Denvi, 2019-11-04:
Hi, looks like #283.
What Candle version you using?

lalalandaudio, 2019-11-04:
Thanks , will try this .
Not sure about Candle version , will check.
Another issue , i can make a heightmap , but not zero probe , i am assuming that using a probe it should be able to probe the z zero ? Unless its for something else ?

Thanks

evildeadhc, 2019-12-10:
I have a similar issue #289.  I described what is happening, but in Russian.  when returning to the operating point, the spindle is deepened by 0.5 mm

closed 2025-09-14 by github-actions[bot]