Denvi/Candle #308: Candle Experimental

iMaks-RS, 2019-11-06:
Предложение: Пока всё  что касается этой версии - писать здесь, в т.ч глюки у 1.1.7, которые желательно устранить.

Глюк, пойман в  версии 1.1.7:
* Если прервать выполнение УП по кнопке "прервать", то потом не работают многие функциональные клавиши, в т.ч пользовательские 1-4. При этот отправка в "дом" работает". Восстанавливается функционал после сброса контроллера grbl. Повторяемость - 100%
* При некорректном выходе из ПО, ("крестик" на окне приложения), станок начинает самопроизвольные движения по осям, иногда с включение шпинделя. Повторяемость - крайне редкая, но место имеет. Версии ОС: Вин7 и Вин10. Вин8 нет, проверить не могу. Поступаю просто - сначала выключаю станок, потом закрываю приложение. Стало привычкой. Сам глюк неприятный, многие новички с ним сталкиваются.

Вот так выглядит на ноубуке Асус X542 интерфейс 1.1.7. 
[image]

Предложение по версии Experemental: 
* Модернизировать функционал консоли, что-бы имела доступ к переменным приложения. Т.е. можно было выполнить команды с заданием переменных, как в функциональных клавишах. Либо сделать отдельно, и расположить как окно с G-Code или картой высот. Потребуется кнопка "выполнить".
* Новая возможность, после введения доступа к переменным, возможно потребует больше кнопок, программируемых пользователем.. 
* продублирую здесь, ранее озвученное - работа с крае-центроискателем "из коробки". Текущая реализация - замечательная, но переменные придётся постоянно менять.. т.е. лазить в настройки.. Мне не лень.. но интерфейс бы не помешал.
* В виду ограниченности функционала прошивки, ввести проверку УП на такой параметр, как выход за пределы. Функция доступна только при заданных софт-лимитах или активных железных лимитов - но с указанием их в самом ПО. При активных софт-лимитах, читать параметры из EEPROM контроллера. Приоритет - EEPROM
* ))интерфейс под 1366*768, что-бы все элементы управления были доступны. Визуализация - для меня не актуальна. На первом место эргономика и выполнение УП. Возможно имеет смысл сделать версию ноутбуков.

Спасибо за Ваш труд!
С Уважением.

dremin1984, 2019-11-06:
Визуализация обязательно нужна!!!!
Есть предложение, добавить 4 ось т.к. на MEGA2560 это возможно.
И сборку для raspberry.

iMaks-RS, 2019-11-08:
Введение горячих клавиш, активировало "фишку", которой очень не хватало - это Ctl+O для открытия файлов  ))). Приятный бонус )

Предложение по версии Experemental:
* Реализация раздельного задания подачи и шага по осям. Либо в настройках, либо в интерфейсе управления. Зачем это нужно?  Иногда возникает необходимость, без создание УП, в ручном режиме обрезать заготовку. При текущем способе - это немного не удобно, но всё-таки возможно, ибо приходится постоянно менять значения шага и подачи по Z, потом опять править шаг и подачу для XY. Может просто добавить это в настройки, активировать или нет этот режим, будет выбирать пользователь на интерфейсе  чекбокосом
* Датчик инструмента - работа с "таблеткой": расположение - стационарно или свободно. Если стационарно, то задавать координаты установки и высоту, если свободно - только высоту. Стационарный режим доступен только при активированном хомлении в контроллере.
* Смена инструмента - в GRBL она не реализована, Если Candle возмёт на себя эту часть управления - было бы здорово. После смены - станок отправлялся бы на стационарный датчик для измерений. Тогда бы не пришлось для каждого инструмента делать свою УП или оптимизировать так, что-бы одним инструментом сделать как можно больше.. (печатные платы делаю кукурузой 0.5-0.8мм, в G2G использую алгоритм разбиения Вороного, отверстия, обрезка и изоляция - всё одной фрезой)
* Хранить рабочее координаты в Candle, ибо странное поведение контроллера. Забиваю G10L2P2коородинаты, но вместо G55, контроллер по #$ выводит их как G54. Тоже самое с p3,p4,p5. (прошивка 1.1f)

XRay3D, 2019-11-13:
Вам необходима помощь по новой версии? Могу помочь, если нужно, сильно шарю в плюсах.

iMaks-RS, 2019-12-16:
Предложение по версии Experimental:
* Идея с веб-камерой классная. но на данный момент не удалось подключить не встроенную в ноутбук, ни внешнюю (Hercules HD). При этом в приложениях к этим камерам - изо есть.
* Добавить возможность старта УП с указанной строки (вводом номера строки, или как в LinuxCNC - по правому клику выходит контекстное меню.. одна из строк "начать с этой строки")
* В случае потери связи с контроллером во время трансляции УП, запоминать УП, и номер строки, где произошла потеря связи. Провёл  опыты, при отключении питания станка. За счёт остаточной энергии в емкостях, МК принимает команды ещё около секунды, после остановки механики.. Т.е фактическая точка остановки механики и потери связи с контроллером - отличаются.. Какое количество кадров успеет проскочить  - х.з, 
* Хотелось бы видеть кнопку, которую по моей просьбе сделал Дамир (XRay3D), [на форуме cnc-club](http://www.cnc-club.ru/forum/viewtopic.php?f=155&t=25727&p=536901#p536901). Это позволит освободить пользовательские кнопки и навесить на них другие функции.
* Либо более расширенный вариант этой "кнопки": добавить две кнопки "установить", "перейти", и выбор из списка координат. Для задания нужной СК, подводим оси в нужное положение, выбираем из списка G54-G59 и жмём "установить" Если нужно сменить СК - выбираем из списка G55 и жмём "перейти", оси поехдут в заданные координаты. [image]

 Здесь стоит исключить возможность наехать инструментом на заготовку или фиксатор.. Поэтому
1.  при перемещениях, сначала ось Z отправлять в G53F500Z0, потом оси XY в заданные положения.
2. выдать пользователю сообщение, что ось Z будет опущена на заданное расстояние и есть шанс сломать инструмент или испортить заготовку. Предложить пользователю выбор "Да", "Отмена", "Измерить длину инструмента"

Ещё раз спасибо за Ваш труд!

dbxzjq, 2019-12-18:
Can you increase the MPG electronic handwheel function and add the fourth or fifth axis, and you can set whether to turn on above the fourth axis

iMaks-RS, 2020-01-29:
Продолжу пожелания в этой теме:
На сколько реально реализовать, (в виде плагина), работу с преобразователем частоты по RS-485 и отображение статуса как LinuxCNC ?  
Сложность в том, что отсутствует какой-либо стандарт на регистры.. поэтому у каждого производителя "свои мыши".  Под LinuxCNC есть [универсальная заготовка](https://www.cnc-club.ru/forum/viewtopic.php?f=15&t=9406#p229460) для работы с modbus, регистры прописываются в ручную, согласно мануалу. Однако для инверторов HuanYang - она не подходит.. но по дефолту в LinuxCNC работы реализована именно с этим инвертором. Если в течении года-двух  появится такая возможность - будет здорово!

closed 2025-09-14 by github-actions[bot]