Denvi/Candle #310: How to install plugins ?

ASHIT-AXAR, 2019-11-15:
how do you install the plugins? (or make a new one?)
there is a default plugin example "test1", but i couldn't find out how to install that ? :thinking: 
does it need to be compiled with Candle ?
any instructions ?

Denvi, 2019-11-15:
Folder "test1" should be placed in "plugins" dir of Сandle app:

```
Candle
    plugins
        test1
    …
    candle.exe
```

There is no need to compile anything except translation files (.qm).

Folder with every plugin must have following:

1. Configuration file 'config.ini' with params:
    - 'name' - name of plugin;
    - 'title' - title of groupbox/dockwindow (translatable);
    - 'type' - 'window' or 'panel' type of plugin.
2. Script file 'script.js' - plugin program, written in JavaScript.
3. Plugin UI file 'widget.ui' - main plugin UI, designed with Qt Designer.
4. Settings UI 'settings.ui' - plugin settings UI. Optional.
5. Translation files 'translation_LL.qm' (LL - locale).

JavaScript program can use following global objects:
1. 'main' - main Candle object.
2. 'mainSettings' - Candle settings.
3. 'ui' - plugin UI.
4. 'settings' - plugin settings UI.

The documentation will be developed later.

zubon2003, 2020-02-17:
I'm enjoying plugin function.
Can I get working coodinate from javascript side?
https://www.dropbox.com/s/jpiq263nrfyv981/test.mp4?dl=0&fbclid=IwAR0TYh_f6NG6B9uBQkhBeJwmL0-IlOaPHO8WVQ-1SxhtUdJkJABp-0T36lQ

Denvi, 2020-02-17:
There is 'vars' object in script environment, has type 'ScriptVars' ([scriptvars.h](https://github.com/Denvi/Candle/blob/Experimental/src/candle/scriptvars.h)).
~~~
vars.Wx
vars.Wy
vars.Wz
…
~~~

zubon2003, 2020-02-17:
Thanks Denvi!
I confirmed it.
And I got it!

alxkos, 2022-10-16:
Hey Denvi, any chance you made a doc for plugin? 
I'm trying to make my own plugin, first created as you described, but not loads. 
Thanks

closed 2025-09-13 by github-actions[bot]