Denvi/Candle #314: Не работает камера

anbod2, 2019-12-23:
Мне не удалось получить изображение. компьютеры разные, ОС разные, Вин7, Вин8, Вин8,1, Вин10. Камеры разные. В других приложениях на этих компах камеры работают.
У моих друзей камера а Candle тоже не работает.
Возможно как то надо настроить плагин?

Denvi, 2019-12-23:
В релизе не хватало одной из библиотек. Включена в версию 1.2.8.1b: https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2.8.1b.zip

После запуска приложения необходимо зайти в окно настроек, ввести имя камеры в соответствующее поле (имена доступных камер отображаются во всплывающей подсказке), установить разрешение камеры, закрыть окно кнопкой "ОК".

[image]

Масштаб и положение вида камеры настраивается в окне "Камера" программы:

- масштаб - с помощью колеса мыши;
- положение - перетаскиванием вида с нажатой левой кнопкой.

Пока имеются проблемы с запуском камеры при загрузке программы, лечится сворачиванием с последующим разворачиванием окна программы.

anbod2, 2019-12-24:
Вот теперь работает, спасибо за разъяснения.
Планируется ли развитие плагина в плане перекрестия, шкал или окружностей?

Denvi, 2019-12-24:
Планируется, по крайней мере, перекрестье (#168).

closed 2025-09-13 by github-actions[bot]