Denvi/Candle #315: Lose Position After Feed Hold

reynolds087, 2019-12-29:
When I issue a feed hold in the middle of a cycle, and then press reset, the position randomly changes. I always lose my origin and unable to make adjustments to my operation because the origin is lost whenever I reset after feed hold.

reynolds087, 2019-12-29:
Using 1.1.7 with a Woodpecker CNC v1.1

Denvi, 2019-12-30:
Hi, in 1.1.7 problems with G54-G59 offsets. Try to zeroing used offset.

reynolds087, 2019-12-31:
[image]

Thank you, but as you can see I was not using the work offsets.

ondras12345, 2019-12-31:
Hi, take a look at https://github.com/Denvi/Candle/issues/283

reynolds087, 2020-01-07:
The machine moves if I abort, but if I pause and then reset, it does not move. After a pause/reset, the work coordinates reset back to the same as the machine coordinates. 

$0=10
$1=25
$2=0
$3=2
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=1000.000
$111=1000.000
$112=600.000
$120=30.000
$121=30.000
$122=30.000
$130=200.000
$131=200.000
$132=200.000

reynolds087, 2020-01-13:
Still have this issue. Is there a way to maintain my G92 coordinates after resetting the machine? When I'm running a cycle I click Pause, and then Abort. After that I have to reset to start moving again, and the G92 resets to the machine coordinates. Do I just need to take note of the coordinates before I reset, and then issue the G92 command after the reset?

ondras12345, 2020-01-13:
You can use the the coordinate systems feature (see [G10 L2](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2), [G10 L20](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20) and [G54-G59](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g54-g59.3)).

Did you check the #283 issue?

reynolds087, 2020-01-13:
Thanks. I will try using G54-G59. You mentioned at the beginning of the post that there were problems with that feature. Can you link the issue or describe it please?

ondras12345, 2020-01-13:
> You mentioned at the beginning of the post that there were problems with that feature. Can you link the issue or describe it please?
@Denvi I'd also be interested in this.

reynolds087, 2020-01-14:
I think I found the problem because when I enter G54 into the console my work coordinates change to reflect the absolute machine coordinates adjusted with the G54 offsets. But I should be able to send a G53 to revert back to the absolute machine coordinates, right? Because when I enter the G53 it doesn't seem to revert my work coordinates back. If I zero out the G54 offset using G10 L2 P1 X0Y0Z0 then it now shows the absolute coordinates matched up to the work coordinates.

reynolds087, 2020-01-14:
One more thing to add, even using the G54 does not solve my problem because when I reset the machine after pause/abort from a cycle, the absolute machine coordinates randomly change.

ondras12345, 2020-01-14:
> One more thing to add, even using the G54 does not solve my problem because when I reset the machine after pause/abort from a cycle, the absolute machine coordinates randomly change.

If you didn't have the homing problems, I'd advise you to setup homing...

ondras12345, 2020-01-14:
> I think I found the problem because when I enter G54 into the console my work coordinates change to reflect the absolute machine coordinates adjusted with the G54 offsets. But I should be able to [...]

Is the machine really not moving in G53 or is this just that the coordinates in Candle are displayed incorrectly?

reynolds087, 2020-01-14:
After resetting the machine it stopped changing randomly. So that is ok I think. In G53 it will move. It's just that the work coordinates and the machine coordinates do not match. My understanding is that they should be the same in G53 modal state.

reynolds087, 2020-01-14:
> 
> Your machine coordinates shouldn't change randomly - it should become 0,0,0 after the reset.

I had to hard reset the board to get it to stop doing that.

ondras12345, 2020-01-14:
> In G53 it will move. It's just that the work coordinates and the machine coordinates do not match. My understanding is that they should be the same in G53 modal state.

When you type `G90 G0 X0 Y0 Z0` to the console, does it move to machine 0,0,0 or G54 0,0,0 (is it moving in the G53 machine coordinate system or the G54 coordinate system Candle shows)?

ondras12345, 2020-01-14:
> > Your machine coordinates shouldn't change randomly - it should become 0,0,0 after the reset.
> 
> I had to hard reset the board to get it to stop doing that.

If you have homing disabled, the current point should become machine 0,0,0 after the reset. I _think_ that if you enable homing and unlock the initial alarm instead of running the homing cycle, the coordinates are calculated from the $130 - $132 settings. It should not be random.

closed 2025-09-13 by github-actions[bot]