Denvi/Candle #320: More control on G-code excution

pedjas, 2020-01-10:
It would be great to heave additional control over G-code execution:

- Executing code line by line would be really helpful for debugging or analyzing g-code.

- Option to execute code starting from freely selected line (or even selecting group of lines and executing just that selection) would also be helpful for both debugging and handling stopped machining or fixing some glitches.

DarrenSchwartz, 2020-04-19:
This is number one on my list.  For instance, despite using the height map, sometimes a section of the board does not get fully milled through.  Instead of re-tracing the whole board with a deeper cut, it would be great to just select a line and start from there with a deeper plunge.  Especially since I don't want to mill all of the traces deeper, just a small section where it didn't go through the copper.  Are there any workarounds to achieve the same thing?  I would love to help, but I'm afraid it's beyond my abilities.

ondras12345, 2020-04-19:
Hi,
you can edit the gcode manually to start from a specific line. Make sure not to delete the commands that set machine modes, such as [G21](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g20-g21) and [M03](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m3-m4-m5). You cannot just execute it from a selected line, because that would most likely cause your tool to hit the material with the spindle turned off.

DarrenSchwartz, 2020-04-20:
> you can edit the gcode manually to start from a specific line.

This is an excellent point.  Thanks for that suggestion.  I could still imagine a way to handle this in the UI that would cover off on the gotchas you mention (for example, a popup dialog asking for or confirming default missing parameters such as spindle speed).  I think it would be valuable feature.  Maybe some day i will look at implementing.

pedjas, 2020-05-17:
I often just need to be able to stop and continue from specific line. However, Darren's point is also valid.

it is not matter of editing g-code. If we go that route then we do not need Candle at all. It is Candle usability issue, and option to better control machine from within Candle.

Candle is excellent software. It just needs some tweaks on user interface side.

Braunfeltd, 2020-05-17:
You can already start from a specific line.  Just have to select the line then hold mouse down on send an option presents itself to start from there.  Only caution is if you have a low z safe it could lower down into your project.  So always best to start on edge and manually set the bit above that spot just to be sure.  I use this all the time fore restarts and or recovery.

closed 2025-09-13 by github-actions[bot]