Denvi/Candle #323: Zeroing machine coordinates

pedjas, 2020-01-10:
It is needed to have an option to zero machine coordinates. 

Machine coordinates stay in machine memory and are more safe from accidental change than work coordinates. That can really help saving some work.

ondras12345, 2020-01-10:
Hi, you should use the **coordinate systems** (see [`G10 L2`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2), [`G10 L20`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20) and [`G54`-`G59`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g54-g59.3)) feature rather than trying to zero the machine coordinates.

I don't think GRBL can zero the machine coordinates without doing a hard reset (not the soft reset [`ctrl-x` command](https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md)). You would have to press the reset button on your machine, and thus cause the machine to make the current position the machine zero.

The coordinate system offsets are stored in the EEPROM and thus should persist between machine reboots, even in the event of a power loss. If you use [homing](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle), you should be able to recalibrate your machine coordinates after the machine boots and use the previously stored coordinate system (offset from machine coordinates) to restore the previous work coordinates.

You may run into some problems, though. The author of this project wrote [a comment](https://github.com/Denvi/Candle/issues/315#issuecomment-569610876) saying there are some problems with coordinate system offsets. I haven't tried it myself yet, so I'm not sure what the issue is.

pedjas, 2020-01-13:
Thanks, I will try that way.