Denvi/Candle #324: Добавьте мануал как собрать в mac os x

ah1102, 2020-01-10:
Добавьте мануал в файл readme на главной странице проекта, как собрать в mac os x. Люди собирают как-то, а инструкции нет. Я сам пользуюсь на мак вашей программой, но старой версией. Хотелось бы обновиться.

PNixx, 2020-04-10:
```
brew install qt5
mkdir build && cd build
cmake ../src -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.14.1 && make
```

ah1102, 2020-04-13:
Спасибо! Сборка прошла удачно: https://www.youtube.com/watch?v=AhFBYnCyvHc

ah1102, 2020-04-21:
Есть один визуальный нюанс. Программа собирается без пиктограммы приложения. Можно ли при сборке добавить пиктограмму для приложения?

negativ72rus, 2020-11-08:
> Есть один визуальный нюанс. Программа собирается без пиктограммы приложения. Можно ли при сборке добавить пиктограмму для приложения?
> 
> [image]

можно же после сборки любой png в качестве иконки назначить.
[image]

closed 2025-09-13 by github-actions[bot]