Denvi/Candle #325: Сломало COMPORT 3D принтера

PNechNET, 2020-01-11:
Попытался подключить к 3д принтеру. Подал команду домой. В итоге comport принтера перестал работать корректно. Подключается на 3 секунды и сразу отключается. Печально, придется менять чип.

ondras12345, 2020-01-11:
Hi, try using a terminal such as [putty](https://www.putty.org/) to connect to the machine. Does it respond?

Did you set the baudrate correctly? If the software you use normally to connect to the machine doesn't set it, it may remain in the state Candle set it to.

PNechNET, 2020-01-11:
На данный момент я поймал баг. Я не уверен, что он относится к этой теме, но, при отправке всех координат на принтере домой (в 0 позицию), comport начинает работать.
Я не могу ссылаться на проблему прошивки, так как после случившегося, я полностью очистил чип и заново установил свою прошивку, которая до открытия данной темы работала исправно.

At the moment, I caught a bug.  I'm not sure if it relates to this topic, but when sending all the coordinates on the printer home (to 0 position), comport starts to work.
I can’t refer to the firmware problem, because after what happened, I completely cleaned the chip and reinstalled my firmware, which worked properly before opening this topic.

Board SKR 1.3 (LPC1768)
Firmware Marlin 2.0.x

ondras12345, 2020-01-11:
Does your printer use GRBL?
Which version of Candle and GRBL do you use?

PNechNET, 2020-01-11:
@ondras12345 My printer doesn't use GRBL. I know it, and did all on my own risk. I write this to warn people to it. Thanks for respond.

closed 2020-01-12 by PNechNET