Denvi/Candle #326: simulation

lianzaozi, 2020-01-12:
Thanks for providing such excellent software. How can candle software perform trajectory simulation without hardware connection? I looked for it in software and didn't find where this function is?

ondras12345, 2020-01-12:
Hi,
you can open the gcode file without having the machine connected.

You should see the visualization of the gcode when you open the file. When you select a command in the code, you can see the part of the toolpath that has already been traveled highlighted in a different color than the remaining toolpath. You can configure these colors in `Service > Settings > Colors`.

pedjas, 2020-01-13:
This is great tip. I also had this issue. If machine is not connected to the PC, option to run code for check is grayed out. I did not realize that if you click on actual g-code line, it will show position and tool path in vusualization screen. 

I still would like to run check simulation without machine.

When we are at that, at my PC check simulation runs lightning fast. It is actually not possible to see any details of tool moving. It finishes before I release Send button. It would be great to have option to set simulation speed.

ondras12345, 2020-01-13:
> When we are at that, at my PC check simulation runs lightning fast. It is actually not possible to see any details of tool moving. It finishes before I release Send button. It would be great to have option to set simulation speed.

The Check mode works by sending [`$C`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) to the machine.

> This toggles the Grbl's gcode parser to take all incoming blocks and process them completely, as it would in normal operation, but it does not move any of the axes, ignores dwells, and powers off the spindle and coolant. This is intended as a way to provide the user a way to check how their new G-code program fares with Grbl's parser and monitor for any errors (and checks for soft limit violations, if enabled).

It is not meant to simulate the machine movement, it just checks for errors. Candle isn't processing anything, it justs [sends the `$C` command along with the Gcode](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L2461) to the machine.

pedjas, 2020-01-13:
> It is not meant to simulate the machine movement, it just checks for errors. Candle isn't 
> processing anything, it justs sends the $C command along with the Gcode to the machine.

Visualization works in check mode.  I use it to check if G-code does not have some quirks. Some CAM can generate strange tool movement so it is good to find it out before actually run the job.

ondras12345, 2020-01-14:
> Visualization works in check mode. I use it to check if G-code does not have some quirks. Some CAM can generate strange tool movement so it is good to find it out before actually run the job.

Yes, Candle does exactly the same thing as when you run the gcode, it's just that the machine completes the commands almost immediately.

As I was trying to explain in the previous post, the check mode is a GRBL feature, not a Candle feature.

pedjas, 2020-01-16:
Yes, but as lianzaozi asked. It is not possible to check g-code paths if there is no hardware attached. 

I guess Candle sends to controller g-code line by line. It might be possible if hardware is not attached to skip sending g-code to controller and behave like it was sent and successfully executed. That would allow user who prepares g-code at PC that has no CNC machine attached to check in Candle how code runs.

Additionaly, it may add intentional delay before sending g-code line so it has effect of slower code executing rate that would also be helpful.

ondras12345, 2020-01-16:
> You should see the visualization of the gcode when you open the file. When you **select a command** in the code, you can see the part of the toolpath that has already been traveled highlighted in a [...]

ondras12345, 2020-01-16:
> Yes, but as lianzaozi asked. It is not possible to check g-code paths if there is no hardware attached.

When you select a command, you can see it highlighted in the visualizer.

Candle cannot check for errors because the error detecting algorithm needs to know the machine settings. It is not possible to for example check if a command would trigger a soft limit if you don't know the dimensions of the workspace and the current position of the machine.

> I guess Candle sends to controller g-code line by line. It might be possible if hardware is not attached to skip sending g-code to controller and behave like it was sent and successfully executed. [...]

That would be exactly the same as manually scrolling trough the program.

closed 2025-09-13 by github-actions[bot]