Denvi/Candle #327: Loud Buzzing & error:9 When Homing

reynolds087, 2020-01-13:
I have the hard limits working now with limit switches, but when I try to do homing, the machine travels to the end of the axis and starts buzzing loudly, but never finds the home position. If I reset to stop the buzzing, it returns an error:9 [MSG:Check Limits].
The switches are wired correctly because the hard limits work. This is on a woodpecker cnc 3018

ondras12345, 2020-01-13:
Hi, take a look at [this wiki page](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle).

Please post your `$$` settings.

Make sure you test all the limit switches, not just one axis. Is this happening on the Z axis (it should home first)?

Take a look at https://github.com/gnea/grbl/issues and https://github.com/grbl/grbl/issues . This is a problem with GRBL, not Candle.

reynolds087, 2020-01-13:
I will open the issue there. All 6 of the switches work as expected when jogging the machine into them, they stop, and I have to reset and unlock the machine to move again. Usually have to do it twice, because it locks a second time when moving off of the switch. I've tested both ends of all three axis with the same results. It's only when I try to home it, then the machine acts like it can't see the switches there and just keeps cranking the stepper motor until I reset.

ondras12345, 2020-01-13:
Link to the new issue: https://github.com/grbl/grbl/issues/1594

closed 2025-09-13 by github-actions[bot]