Denvi/Candle #328: How about zeroing X and Y using probe same as for Z?

pedjas, 2020-01-13:
I stumbled on one YouTube video where I noticed operator used aluminum block for probe zeroing all three axes.

Aluminum block is set so its corer is preciselz at corner of the material and his software has option not just to Z-probe but also X-probe and Y-probe by touching sides of the aluminum block. It looks as really cool and useful feature.

Can this be done with Candle?

ondras12345, 2020-01-13:
You can achieve this by typing gcode directly to the Console. Take a look at the [`G38.n` commands](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38). You can see the command Candle sends when you press the Z-probe button in `Service > Settings > Probe commands`.

pedjas, 2020-01-13:
I meant to have user interface (buttons) for this, same as for Z-probe

ondras12345, 2020-01-14:
You can set one of the user buttons to do this.

@Denvi it would be a nice feature to have the ability to change the icons of the user buttons.

ASHIT-AXAR, 2020-01-15:
do we need some extra control buttons ?
- a find/goto job XY center button (probing at corner is not the suitable option always)
        or a mini popup window to quick select/goto a corner (X,+, center)
- a goto work_zero button 
- a park button (machine goes to a predefined coordinate before shutdown)

pedjas, 2020-01-16:
User buttons are not quite suitable for this.

ondras12345, 2020-01-16:
> User buttons are not quite suitable for this.

I think this is a rarely used feature that doesn't need a button in the UI dedicated to it (users with small screens would see even less controls without scrolling).

In my opinion, this is a great use for the user buttons - if you find yourself using this feature regularly, you can map it to a user button. I think it is not a good idea to add a dedicated button for something most users will never use.

Keep in mind that machines capable of milling metal are usually not controlled by GRBL, and thus cannot use Candle. The beginner CNCs that are controlled by GRBL are mostly used to mill PCBs and other materials that are generally not condutive and thus cannot be probed from the side.

I really don't think many people use this feature so often that they would need a special button for it.

pedjas, 2020-01-18:
By not suitable for user buttons I meant it is just few of them and user cannot set icon or caption for the button so function is obvious.

To do an x and y probing there is no needed to use metal material. I planned to use simple aluminum cubical for that (or short aluminum rectangular pipe). If I align one corner of the cubical with zero point on material and align x and y axes then I can simply do x, y and z probing. 

Hobby users usually have to make their own probe for this purpose. Instead of using metallic sheet they just have to use cubical and get probe for all three axes. I see it not more complicated than making probe for just z axis.