Denvi/Candle #330: Please add tool change command support.

Hamulus, 2020-01-16:
GRBL does not support M6 tool change command. However Candle processes G-code before sending it to machine. So it would be good to interpret M6 in incoming G-code as spindle stop, then up Z-axis to some value (it should be defined in settings) and set pause.
So we cound perform manual tool change and then resume G-code sendig since then.

reynolds087, 2020-01-16:
Would love to see this added, as well as custom variables.

pedjas, 2020-01-19:
I just came to post the same proposal. So I am joining in.

MarkSDS, 2020-03-10:
WIth FlatCam generated Gcode files, these toolchanges already work like a charm!
 
Alternatively you could try to add toolchanges manually in your Gcode files:
look for any `G00 Z<safeHeightVar>` or `G01 Z<safeHeightVar>`, count these events and add a couple of lines at appropriate places: 
```
G00 Z<toolChangeHeight> 
M06
```

pedjas, 2020-03-10:
GRBL does not recognize tool change command, thus if that is contained in g-code Candle just pops up error message. All you can do is click to ignore and continue.

What is needed is Candle to make a pause, let user jog spindle in safe space, change tool, reset Z and then continue work.

MarkSDS, 2020-03-11:
OK, IC: Z-leveling option is not availlable in the scenario that FlatCam produces. 

Please, somebody, develop the feature ;)

nonpds, 2021-01-18:
I have g-codes in my program that work fine for tool changes but cannot figure out how to reset the Z axis after the tool change. Any help would be appreciated.

larsskj, 2021-04-18:
I would love to see this work as well.

closed 2025-09-13 by github-actions[bot]