Denvi/Candle #331: cnc3018 . when i sent gcode to machine it be come smaller than the real size to smaler

mbalakash, 2020-01-16:
cnc3018 . when i sent g code to machine it be come smaller than the real size i used easel to making the project i think i tried many things it same issue 

cnc card is cronosmaker 
windos 10
candle 32

mbalakash, 2020-01-16:
version 1.1.7

ASHIT-AXAR, 2020-01-17:
Reasons are:
- Machine Units and Job Units are not the same (Metric or Imperial).
- Drivers are not calibrated.

mbalakash, 2020-01-18:
How i do it

ondras12345, 2020-01-18:
Open your gcode and try to find one of these commands: [`G20`, `G21`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g20-g21).

There is also a GRBL setting [`$13`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#13---report-inches-boolean) that sets the units your machine reports back.

And there's also a setting in Candle 'Machine information: Units' that should match the `$13` setting.

mbalakash, 2020-01-19:
Thank you so much I'll try it to day

mbalakash, 2020-01-19:
The probe commands is :G21G91G38. 2Z-30F100;G0Z1;G38.27-2F10
and the safe position commands is : G21G90;G53G0Z0
and i sent the $13 
but still same problem

pedjas, 2020-01-19:
Maybe machine is not calibrated.

Set head to edge to be close to machine construction. Find point where you may easily measure distance form head to construction.

Then send command to move head for specific distance (10 cm, 20 cm..). Measure distance of the head from the same spot on construction and see what actual distance head traveled. If it does not match to the command it is not calibrated.

ondras12345, 2020-01-19:
> The probe commands is :G21G91G38. 2Z-30F100;G0Z1;G38.27-2F10
> and the safe position commands is : G21G90;G53G0Z0

I wanted you to find the G21 command **in the g-code program file** you are running, so that we can determine whether your job is in inches or mm.

I also wanted you to check the `Machine information: Units` setting in Candle. Probe commands and safe position commands are not involved in this problem.

> and i sent the $13

I'm not sure what this sentence means. Did you issue the `$$` command to see the settings?

ondras12345, 2020-01-19:
> Then send command to move head for specific distance (10 cm, 20 cm..). Measure distance of the head from the same spot on construction and see what actual distance head traveled. If it does not match to the command it is not calibrated.

Reset the machine and type `G91 G0 X1 G90` to the console. This should move +1 (mm or ") relative to current position on the X axis.

Measure the distance it traveled. If it is 1mm, then your machine units are mm. If it is 25.4mm, your machine units are inches.

If it is something else, you may have a calibration problem. Post your `$$` settings in this case.

mbalakash, 2020-01-19:
> Reset the machine and type `G91 G0 X1 G90` to the console. This should move +1 (mm or ") relative to current position on the X axis.
>
> Measure the distance it traveled. If it is 1mm, then your [...]

I sent it in console and >error:21

mbalakash, 2020-01-19:
> I wanted you to find the G21 command **in the g-code program file** you are running, so that we can determine whether your job is in inches or mm.
>
> I also wanted you to check the `Machine [...]

[image]
Did you mean that

ondras12345, 2020-01-19:
> error:21

Ok, GRBL needs the commands separately:
```
G91
G0 X1
G90
```

ondras12345, 2020-01-19:
> [image]
> Did you mean that

Yes, your Job units are mm then. What factor do the dimensions get reduced by when you actually run the job? (The result should be the same as with the test commands I sent in my previous post.)

mbalakash, 2020-01-19:
[image]
That to opjects by same gcod and (mm)

ondras12345, 2020-01-19:
This is most likely a calibration problem. Try to determine the reduction factor and post your `$$` settings.

mbalakash, 2020-01-19:
> This is most likely a calibration problem. Try to determine the reduction factor and post your `$$` settings.

mbalakash, 2020-01-19:
[image]
..

mbalakash, 2020-01-19:
[image]
2

mbalakash, 2020-01-19:
[image]
3

mbalakash, 2020-01-19:
[image]
4

mbalakash, 2020-01-19:
[image]
End

mbalakash, 2020-01-19:
> This is most likely a calibration problem. Try to determine the reduction factor and post your `$$` settings.

Isent pictures of $$ settings can you till if any thing goes wrong

ondras12345, 2020-01-19:
The problem _may_ be the `$100`-`$102` settings. The [first google search result](https://forum.arduino.cc/index.php?topic=627389.0) says it should be 800.

mbalakash, 2020-01-19:
Ill try it

mbalakash, 2020-01-19:
It gose worse  

> The problem _may_ be the `$100`-`$102` settings. The [first google search result](https://forum.arduino.cc/index.php?topic=627389.0) says it should be 800.

mbalakash, 2020-01-19:
Whin I clicked at send it terns of and it fixed the size issue

mbalakash, 2020-01-19:
I am trying to send video for it

mbalakash, 2020-01-19:
It fixed thanks everyone
I used $3=5

closed 2020-01-19 by mbalakash