Denvi/Candle #332: Console panel cannot shrink

pedjas, 2020-01-25:
I use Candle 1.2.8b.

Last night I exited application as usual. This morning when I fired it up Console panel became too wide and forced minimum size so I cannot resize it.

I went through configuration but found no option to set that. I searched through settings.ini but did not find any setting that looks like controlling this. There is setting for Console min height but not width.

However, when I deleted settings.ini and restarted program, it recreated all panels so Console panel went back to normal (however, all other settings were bit messed up, even with resetting default settings.)

Denvi, 2020-01-25:
Hi, all windows controlled by "formMainState" setting.

pedjas, 2020-01-25:
I tried deleting it and it did not change a thing.

closed 2025-09-13 by github-actions[bot]