Denvi/Candle #333: Candle does not always visualize loaded g-code

pedjas, 2020-01-25:
Candle 1.2.8b

Sometimes, it happens when I load g-code into Candle it does not visualize it. I could not find some pattern when this behaves. It happens with both generated g-code or manually written.

G-code is loaded, everything else works: it may be sent to machine, even processing is animated in visualization - cutting tool runs but on the empty screen.

Example of code that is not visualized is in attachment.

ondras12345, 2020-01-25:
Your gcode doesn's specify the Z coordinates.

You already have `G92 X0 Y0 Z0` in your code, so you can just add `G0 Z0` after the `G90` line. The machine shouldn't move because the `G92` command made the current point 0,0,0.

pedjas, 2020-01-25:
Adding Z0 did the trick for this example (this code was used to test laser so setting Z was not needed). Thanks.

But this happens even with some complex G-code for milling with multiple passes, where Z is used. Next time it happen I will save g-code to show.

ondras12345, 2020-01-25:
> But this happens even with some complex G-code for milling with multiple passes, where Z is used. Next time it happen I will save g-code to show.

I haven't encountered that yet. Are you sure the code contained **absolute** Z coordinates? It could have used the [`G91` incremental mode](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g90-g91).

closed 2025-09-13 by github-actions[bot]