Denvi/Candle #334: Карта высот

vladinen, 2020-01-25:
Не могу понять. 
1. Делаю измерение длины инструмента Z щупом. Обнуляю рабочую координату по Z (при этом машинная координата ну скажем "-20").
2. Сканирование сетки поверхности в итоге выдает все по нулям (хотя в консоли видно что машинные координаты показывают  действительную кривизну поверхности, ну скажем от -19.5 до -20.5.
Вопрос как машинные координаты привязать к рабочему нулю?
И второй вопрос. Почему зондирование поверхности не начинается именно с указанной рамки, а первый замер производит в X0 Y0, ведь там нет заготовки?

Denvi, 2020-10-01:
Hi, your firmware has 4 axes in probe response, Candle works only with 3 axes for now.

kirovreporting, 2020-10-04:
Thanks! I disabled A axis in firmware and it worked)

closed 2025-09-13 by github-actions[bot]