Denvi/Candle #337: GCODE Extension

nicola-miller, 2020-01-28:
Would it be possible to add the '.gcode' extension to the list of native files that can be read by Candle?

Files with this extension are produced by Vectric software and are the most suitable for controlling my CNC machine through Candle. 

It would be convenient if this type of file could be 'dragged' into Candle, or just read from <File Open>, instead of having to select through <All Files>.

Thank you.

ASHIT-AXAR, 2020-01-30:
alternative way:
open vectric's post-processor data folder:
"C:\ProgramData\Vectric\Aspire\V9.5\PostP"

now open the post-processor file you are using. search for (FILE_EXTENSION = "gcode")
change it to something you desire. (for ex: "gcode" to "tap")

nicola-miller, 2020-01-30:
Brilliant. Thank you so much for a simple solution to an irritating problem.
All works properly now.