Denvi/Candle #339: Spindle controlled with servo signal

Danccls, 2020-02-02:
Hello comunity, I'm testing this Grbl version https://github.com/cprezzi/grbl-servo because on my mini CNC I want to change my poor 555 spindle motor by a RC brushless and a ESC, to have more RPM.

The system look great (now I have a standard servo to check it) but I have a little trouble. At start I must send the M3 Sxx (I do it by an user button), where xx are the RPM I want. But those RPM are between 0 and 255. I don't mind the relaction between the maximun value (255) and the real RPM.

On the program, once I selected the RPM, I can change it in the Spindle box, but only tree values: 0,100 and 200, because the step to change the RPM are 100.

My question is: There is any way to change it and select steeps of 2, 5 or 10 on the spindle speed selector?

I can do it on the command line with the M3 command, but it will be better to do it by the grafhic interface ;-) 

Thanks a lot and very good job!!

Danccls, 2020-02-02:
By the way, I use the 1.1.7 and the 1.2.8b versions

ondras12345, 2020-02-02:
Hi, the value seems to be fixed. [src/frmmain.cpp#L382](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L382) [src/widgets/sliderbox.h](https://github.com/Denvi/Candle/blob/master/src/widgets/sliderbox.h)
It shouldn't be hard to make it configurable, though. You can try to implement it yourself.

ondras12345, 2020-02-02:
Alternatively, you could set [`$30`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm) and Spindle speed max (Candle settings) to something like 10000.

Danccls, 2020-02-05:
Hi, thank you for you answer.

I'm a hardware technician, not a software developer and I still can't implement it, but I will try to get it.

Regarding the change of the $ 30 parameter, in this version of Grbl the maximum speed should be 255. Greater than this I have still checked if it works. I'm on a work trip and I will not be able to check it for a few days.

Thanks for the support, Dan.