Denvi/Candle #341: Incorrect "Height map" algorithm for 4 or more axes.

vitabu, 2020-02-04:
When using the grbl branch for 4 or more axes, the "z" coordinate in the "height map" algorithm is incorrectly parsed. The height map is filled with zeros. This is because the regular expression in ". * PRB: ([^,] *), ([^,] *), ([^] ^:] *)" in cap (3) takes everything up to the character ":" . You need something like ". * PRB: ([^,] *), ([^,] *), ([^] ^: ^,] *)".

henpemaz, 2020-03-20:
I've had the same issue, I'm using [grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X/) for my 4-axis machine.

What I had to do was temporarily flash a vanilla GRBL firmware in order to get it working for a PCB job, and then switch back the firmware when I was done... That, or probe manually and fill in the values which got old super fast.

henpemaz, 2020-03-26:
I am too lazy to download and set up a build environment for this project, so I opened the executable file, found and edited the regex string.
`
before:  
([^]^:]*)
after: 
([^],:]*)`

And it worked like a charm. That second `^` was redundant since the first one already states that the set is exclusion. It's very important that it has the same length after edited.

Editing executable files is dangerous. Try at your own risk!

eightheads, 2021-03-29:
Thank you for this, it was driving me absolutely crazy! And thanks @henpemaz, I was able to edit the executable on my Mac to fix the issue. Seems like a quick fix.

Slemi, 2023-02-22:
> I am too lazy to download and set up a build environment for this project, so I opened the executable file, found and edited the regex string. ` before: ([^]^:]*) after: ([^],:]*)`
>
> And it worked [...]

Any chance you can let us know step-by-step how to do this on windows?