Denvi/Candle #342: Milling does not complete

vj010489, 2020-02-06:
Hi 

so i brought a cnc machine the 3018 from amazon and it came with the software grblcandle
i tried it say 3 times trying to get a feel for the machine, i will mainly use it for milling pcb's. so after a few attempts i got it to work perfectly and i milled a really good pcb. now the problem i am getting is that i haven't used it for a few days, so today when i finally looked at it again and ran the same gcode i did for the perfect pcb, the program just hangs, it starts going great however a few mins in or whenever the milling machine just freezes. i keep getting a serial port error 11, saying a device attached to the machine isn't working. please can someone help me solve this issue, as i am starting to think i wasted my money. i have tried candle v1.2.8b and 1.1.7, i am really hooping someone helps me here 

Thank you

ondras12345, 2020-02-07:
Hi,
try a different USB cable. This problem may be caused by electromagnetic interference from the stepper motors. A cable with a ferite bead might help. Also check if the connectors on your power brick are making a good contact.

Error 11 is an UnknownError: [frmmain.cpp#L1369](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L1369) [QT docs](https://doc.qt.io/qt-5/qserialport.html#SerialPortError-enum).

The control board most likely uses the CH340 USB to UART converter chip, which is known to work unreliably. You can try googling some troubleshooting advice for it.

If you are using a laptop, try disconnecting it from the mains to prevent ground loop problems.

vj010489, 2020-02-14:
Ok figured it out if anyone else is having this issue. i'm using a copper clad board with a metal clamp, couldn't believe i didn't notice that, so it caused ripples in the power. all i did was connect a crocodile clip from the metal clamp holding the board to a GND pin on the CH340 USB to UART controller and boom it worked 

Thank you

samehherit, 2020-03-20:
Hi / Vj010489
I have suffered from this problem a few months ago 
The problem was with the usb cable 
I replaced it with a shielded  usb cable and the machine stopped hanging 
Good luck