Denvi/Candle #349: PROBLEMS WITH JOG, ZERO XY, ZERO Z DONT WORK IN WIN10

AugustoBRAZIL, 2020-03-25:
Dear in win10, candle 1.2.8.1b does not work the jog buttons, on the console it shows for example: $ J = G21G91X1Y0Z0F100 <error: Bad number format, on the control panel the origin button also does not work shows the error $ H <error : Setting disabled, the ZERO XY and ZERO Z buttons also do not work and do not update in the state, could someone help me with these problems?

AugustoBRAZIL, 2020-03-26:
[image]

AugustoBRAZIL, 2020-03-26:
[image]
[image]
[image]

ondras12345, 2020-03-26:
Which version of GRBL is your machine running? It si displayed in the console when the machine starts.
See https://github.com/Denvi/Candle/issues/252#issuecomment-433629101

AugustoBRAZIL, 2020-03-26:
I'm using Grbl 0.9i, with Mana controller v2.5, can I upload another updated version of GRBL?

AugustoBRAZIL, 2020-03-26:
I updated the version to GRBL 1.1 and it works perfectly. Thanks for your help
Other questions and proposals to further improve the software:
1-Could you include a camera crosshair in the next software update?
2-Could you include a plugin for collecting points coordinates?
3- Could you include a function or a plugin for basic 2D measurements using the points collected as: horizontal line (in X direction, with 2 points), vertical line (in Y direction, with 2 points), line in any direction (with 2 points), circle by 3 points, angle by 4 points (2 lines)?

AugustoBRAZIL, 2020-03-28:
Would anyone have Denvi's email contact?