Denvi/Candle #350: Candle Supporting resuming ?

samehherit, 2020-03-26:
Does candle supporting resuming work incase of the power breakdown ? 
If the answer is yes , then how?

ondras12345, 2020-03-26:
The machine coordinates and settings are kept by the machine (GRBL firmware), not Candle. If your machine loses power, Candle cannot do anything to recover from the coordinate loss.

However, you can home your machine in order to recover from the power loss. That way, your machine coordinates will have the same origin as before the power loss.

For the coordinate offsets you most likely use to make the corner of your stock the 0,0,0 point, see https://github.com/Denvi/Candle/issues/202#issuecomment-450739964 and https://github.com/Denvi/Candle/issues/323#issuecomment-573120283 , or just write them down on a piece of paper.

After recovering the coordinate system, you can manually edit the gcode to resume from a specific line or just run the same gcode again if you don't mind having to wait for the machine to reach the same cut depth as before the power outage without actually milling anything. If you decide to edit the gcode, make sure to not delete the commands that set machine modes, e.g. G21 or M03. You cannot just execute it from the line it stopped at, because that would most likely cause your tool to hit the material with the spindle turned off.

samehherit, 2020-03-26:
Perfect answer 👍🏼
Thanxx alot

TIPLDN, 2020-06-09:
Hi there,

So I gone to test it now and managed to sorted the USB cable issue out. When I send the program the mill hit dragged through the acrylic, yesterday it lifted to a safe hight? Why is this?

TIPLDN, 2020-06-09:
Also I see I’m getting a unsupported command shall I Delet that line?

Braunfeltd, 2020-06-09:
> The machine coordinates and settings are kept by the machine (GRBL firmware), not Candle. If your machine loses power, Candle cannot do anything to recover from the coordinate loss.
> 
> However, you can home your machine in order to recover from the power loss. That way, your machine coordinates will have the same origin as before the power loss.
> 
> For the coordinate offsets you most likely use to make the corner of your stock the 0,0,0 point (work coordinates), see [#202 (comment)](https://github.com/Denvi/Candle/issues/202#issuecomment-450739964) and [#323 (comment)](https://github.com/Denvi/Candle/issues/323#issuecomment-573120283) , or just write them down on a piece of paper.
> 
> After recovering the coordinate system, you can manually edit the gcode to resume from a specific line or just run the same gcode again if you don't mind having to wait for the machine to reach the same cut depth as before the power outage without actually milling anything. If you decide to edit the gcode, make sure to not delete the commands that set machine modes, e.g. G21 or M03. You cannot just execute it from the line it stopped at, because that would most likely cause your tool to hit the material with the spindle turned off.

Best practice after homing and resetting your Zeros with out the need to edit the code is to find what line is closest to where you left off by clicking on it until you see the grey dot on the visualizer where you where.  than find the line in the gcode a few steps back at the edge of the work as you want to start on an edge to ensure that the Z does not plunge down into the work I have seen if you start in middle where the bit went down getting to that spot.   Next once you have the grey dot at the edge move your spindle over top of it start the spindle up.  than click and hold on the Send button a pop up of resume from this spot comes up.  click to proceed it will than start at that spot with no issues.  This way you dont have to play with any of the code.   hope that helps :)

TIPLDN, 2020-06-09:
Thanks. I have managed to resolve the lost connection issue, now having the mill bit drag after homing Zero XY

Braunfeltd, 2020-06-09:
> Thanks. I have managed to resolve the lost connection issue, now having the mill bit drag after homing Zero XY

is the Z no lifting when you home?  if not than you have something not right with your Z axis as it should lift up on a home until it hits limit switch.