Denvi/Candle #352: CNC3018 spindel motor not running when sending Image for engraving

RLBeadon, 2020-03-29:
A very good day to you all. 

I have recently purchased a3018 CNC engraving machine from Banggood.

I have assembled it and am now readt to test it.

I generated the gcode of an image using Easel and opened it up in grblControl.

I can Jog the machine left and right, forward and backwards and spindel motor up and down.

I can turn the spindel on and off via the software.

However I am facing 2 Challanges

1. If i need to do an emergancy stop I cant, for example if I am jogging the spindel motor up and it goes to high how do I stop it immedietly?

2. When I send the code for engraving  the spindel motor will not start automatically. I have to abort the process, reset, start spindel manually and then send.

I am very new to this having only been doing it for 2 days, I am not sure if there is someting I should have done in the set up or I have missed when generating the gcode.

Your assistance will be greately appreciated 

Many thanks

Rodney

ondras12345, 2020-03-29:
Hi,
You need to set **Automatic spindle control** in Easel (Machine menu).

The **Reset** button should stop the machine immediately. However, you'll probably loose the current position. See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands
```
0x18 (ctrl-x) : Soft-Reset

Immediately halts and safely resets Grbl without a power-cycle.
Accepts and executes this command at any time.
If reset while in motion, Grbl will throw an alarm to indicate position may be lost from the motion halt.
If reset while not in motion, position is retained and re-homing is not required.
An input pin is available to connect a button or switch.
```

RLBeadon, 2020-03-29:
@ondras12345  thank you very much for your assistance, it is working perfectly now. Thank you

closed 2020-03-29 by RLBeadon