Denvi/Candle #354: CNC 1610 engraver stops half way through

dswest1, 2020-04-04:
I'm new to this, so be gentle. I have a CNC 1610 that starts going through the program, but stops randomly before it's finished. The spindle increases speed when it stops moving and I can not get the spindle to stop with the program, Candle. The only way I get it to stop is to turn off the cnc machine. Is this a software issue or is the machine junk ?

ondras12345, 2020-04-04:
Hi, this may be a hardware issue. See https://github.com/Denvi/Candle/issues/342

Does Candle show any error messages? Check the console. Does the machine still respond to commands with `ok`?

dswest1, 2020-04-05:
Hi,
No error messages show, but the g-code scrolling down the bottom stops when the machine does. There is no way to stop the machine through the program. I have to switch the machine off to stop it.
Settings below

$$ < $0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=1
$11=1.000
$12=0.002
$13=1
$20=0
$21=1
$22=1
$23=7
$24=25.000
$25=500.000
$26=250
$27=2.000
$30=1000
$31=0
$32=0
$100=500.000
$101=500.000
$102=500.000
$110=8000.000
$111=8000.000
$112=500.000
$120=500.000
$121=500.000
$122=10.000
$130=500.000
$131=500.000
$132=200.000
ok

ondras12345, 2020-04-05:
Ok, that seems to be a hardware issue. The machine stops communicating with your computer.

Try wiring a 'Reset/Abort' switch to your controller board (https://github.com/gnea/grbl/wiki/Connecting-Grbl and https://github.com/gnea/grbl/issues/123#issuecomment-279131776) to see if it is a GRBL issue or communication issue.

In case the machine stops after you press the button in the error state, the GRBL firmware is OK  and you should focus on the communications hardware. Read the link from my first post.

svenhb, 2020-04-05:
If you use Arduino, you may exceed the 30 kHz step frequency: https://github.com/grbl/grbl/issues/41  
And https://github.com/grbl/grbl/issues/1332   
$100=500; $110=8000 - results to 66.6 kHz  
I had similar problems with bare boards on desk to check my grbl-sender... 
Suddenly Status message showed ```<Run|WPos:3.011,4.536,2.412|Bf:51,128|FS:0,0>```  
Bf:**51**,128 is not possible...

dswest1, 2020-04-07:
So I need to change the settings to change the frequency then ? Or do I try a usb cable with a ferite block on both ends ?

svenhb, 2020-04-07:
Yes, try $110, $111 = 3500 mm/min to stay below the 30 kHz...
It's a problem of the controller, not of the USB cable.

ondras12345, 2020-04-07:
> Yes, try $110, $111 = 3500 mm/min to stay below the 30 kHz...
> It's a problem of the controller, not of the USB cable.

It could be a cable problem, the CH340 ICs are very intolerant to interference. I didn't know about the 30 kHz limit, though.

dswest1, 2020-04-07:
Well.....thank you both for your help. Changing that setting has solved that issue. Now I just have to sort out why it's smaller than the dimensions I setup in the file. If that makes sense....... cheers.