Denvi/Candle #356: Grbl Settings not changing

RLBeadon, 2020-04-07:
Good DAy

I have been trying to change the direction of my X axis movement.

The Y & Z axis are in the right direction

I have entered $2=1 which should invert the x axis and therefore is direction of travel. 

However after entering this nothing changes X,Y,Z  all travel in the same directions. I have restarted the software and have the same problem.

Please may I have some advice

ondras12345, 2020-04-07:
Hi, you need to change $3, not $2.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#2--step-port-invert-mask
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

RLBeadon, 2020-04-07:
@ondras12345  Thank you for pointing my error to me. 

I changed it according to the table

Firstly X,Y, needed to change direction but not Z so entred $3=3

When I tested again X,Y did not need to change Direction But Z Did so I entered $3=4

This inverted everything back with X and Y being in the right direction but Z not.

I checked on the PCB control card and the cables are plugged into the zight ports as per the markings on the board

Totally confused now

ondras12345, 2020-04-07:
Maybe you already had some axis inverted before you started changing the settings.

RLBeadon, 2020-04-07:
@ondras12345 , I dont know how to post my settings, or where to find a list of them

ondras12345, 2020-04-07:
I've updated my previous comment. If you want to know more about GRBL settings, read this wiki page: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

MeJasonT, 2020-04-07:
Are you moving by jogging or by homing
you can have homing move the axis in a totally different direction to jogging.
I imagine from what im reading is that you are setting up the axis move (jogging) directions.
As you have already figured out the axis are Z Y X with binary coded decimal BCD values of 4 2 1
Z reversed and XY normal would be 4+0+0 =4  therefore $3=4.  if you still have problems you can always swap the wiring pairs for the motor so A+ A- B+ B-  would become B+ B- A+ A-.  If you have the arduino IDE and connect the serial monitor from within the IDE to GRBL, send $$ and it will list all your settings, other port monitor applications should be able to do the same.  Its a bit of a problem with GRBL gui's that they dont offer a facility to export settings.  Its not as if the settings cant get lost or become erased from Eprom (internal memory).

RLBeadon, 2020-04-08:
> Maybe you already had some axis inverted before you started changing the settings. Post your settings, please.
> Edit: Type `$$` to the console to list your GRBL settings.
> 
> > .. so I entered $3=4
> > This inverted everything back with X and Y being in the right direction but Z not.
> 
> Try $3=0
@ondras12345 
I have copied and pasted below the settings

$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=1 (step port invert mask:00000001)
$3=4 (dir port invert mask:00000100)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

RLBeadon, 2020-04-08:
@MeJasonT  Thank you for your reply I have answered the questions you asked in the Body of the text

> Are you moving by jogging or by homing   " I am Jogging "

> you can have homing move the axis in a totally different direction to jogging.  " When I press home nothing happens, the following is the message on the consoul $H < error: Setting disabled
S2000 < ok"
> I imagine from what im reading is that you are setting up the axis move (jogging) directions. "Yes that is correct , I want to use the screen buttons to move up down, left right, back forward but this is not happening"

> As you have already figured out the axis are Z Y X with binary coded decimal BCD values of 4 2 1
> Z reversed and XY normal would be 4+0+0 =4 therefore $3=4.  "I tried this and it only changed the Z axis which is now  correct"

if you still have problems you can always swap the wiring pairs for the motor so A+ A- B+ B- would become B+ B- A+ A-. If you have the arduino IDE and connect the serial monitor from within the IDE to GRBL, send $$ and it will list all your settings, other port monitor applications should be able to do the same. Its a bit of a problem with GRBL gui's that they dont offer a facility to export settings. Its not as if the settings cant get lost or become erased from Eprom (internal memory). " sorry I read this several times and its way past my limited knowledge at present"

Thank you for your assistance

ondras12345, 2020-04-08:
Have you already solved your issue? Please close this issue if so.

If not, try this:
- Set `$3=0`
- Determine which axes need to be inverted. If your machine moves the bed instead of the spindle on the Y axis, make sure you know in which direction it is supposed to move. Watch the spindle move relative to the bed, not the opposite. I.e. the machine bed should move towards you when you press the Y+ button (8 on the numpad - press scroll lock to enable keyboard control).
- Because you started with $3=0, you can now just use the [lookup table](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#2--step-port-invert-mask) and set $3 to the right value.

MeJasonT, 2020-04-08:
@ondras12345 we are not sure which coordinate configuration he is using, your suggestion is for negative space/negative coordinate system.  If he sets his machine up with zero in bottom left corner for a paper styled set up as opposed to a profesional CNC miller setup as designed into GRBL by default.  There never really is much explanation regarding this setup, you have to trawl through dozens of issue posts to find an explanation of it.  many fall fowl of this when they expect the machine to do exactly what was on the layout in CAD.
@RLBeadon from the above comment are you using the default grbl setup or trying to make it a page setup.

ondras12345, 2020-04-08:
> @ondras12345 we are not sure which coordinate configuration he is using, your suggestion is for negative space/negative coordinate system. If he sets his machine up with zero in bottom left corner [...]

He is not homing the machine, so he can set the 0,0,0 point manually using offsets. I _think_ you are only talking about origin position.
Aren't the axis **directions** always the same? See [this image](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time).

MeJasonT, 2020-04-08:
both example you linked are for the negative coordinate system (the normal standard way)  some users with routers have inverted their axis so that what they see on paper is how the bed is positioned.  If i move to x100 y100 and set its location as 0,0 
 the work is still carried out from the top right of the work piece taking the work piece away from me and to the right.  Thats not how it would work if it was set to work like it was A4 paper with the 0,0 set in the bottom left corner with X & Y going left and moving the machine towards.  As you noticed with your setup in Fusion 360.

RLBeadon, 2020-04-08:
@MeJasonT , @ondras12345 

From the link sent to me I have set all the configurations to 

$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=1 (step port invert mask:00000001)
$3=4 (dir port invert mask:00000100)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

I then do a test 

I am wanting to, **( X Axis)** push the Jog left button to make the motor move to the left, push Jog right to get the motor to move right  **(Y Axis)** Jog forward the tray moves forward. Jog Back the tray moves backwards.**( Z Axis)** Push up the motor must go up, push down the motor must go down

I have started with the configuration as shown above.

X | Y | Z |  
-- | -- | -- | --
N | N | Y |  
y | y | y | $3=7

So the result is that it is now working 100% how I wanted it to make sure the motors Jog in the right directions  Thank you for all your Help, I have several other challanges but let me try to resolve them myself first.

I will close this just now

ondras12345, 2020-04-08:
Do you really want to have $2=1? I think you set it accidentally according to your first post.

RLBeadon, 2020-04-09:
@ondras12345  changed $2=0

Thank you for all your help

closed 2020-04-09 by RLBeadon