Denvi/Candle #357: Can't reliably probe heightmap at X,Y=0,0

mjml, 2020-04-07:
I can't the auto-heightmap prober because my PCB bottom left edge is at X,Y=0,0. This is because I carve a wasteboard to seat the pcb blank, and so the pcb edge is right at this origin. The probe at this location will sometimes fail, or worse it can displace my copper blank if the copper does not extend right to the edge of the plastic.

When I click on Probe (the button that replaces Send in edit Heightmap mode), that one initial plunge at the XY origin will sometimes fail, resulting in the bit sinking into my wasteboard and missing the PCB copper. Is there a way that I can disable that initial probe? I set the borders of my heightmap to be inset from the edges of the PCB so that I am not probing a "knife edge" of copper. I currently create the entire 6x4 heightmap manually. But I would like to create a higher resolution heightmap using the autoprober.

Can you tell me where in the code I can find this XY origin probe? I would like to experiment with disabling it, also with creating a new checkbox for it. I do not need to probe outside the bounding box. Thank you.

mjml, 2020-04-15:
I found where the probe commands are issued. They are in updateHeightMapGrid() within frmmain.cpp.

loops, 2020-04-20:
Just want to add support for this issue.   When you define a bounding box for the heightmap operation, no probe should be done outside that box at all, including the one currently done every time at 0,0.

The whole point of creating a bounding box is to tell the operation where it can safely probe automatically.  So many times I have a mounting clamp at 0,0 and this means the entire thing has to be mapped manually simply because Candle does this forced probe at 0,0.

*Please* remove this aspect of the heighmap operation, or at the very least make it optional.

mjml, 2020-04-20:
I've implemented a checkbox for the initial probe at 0,0. It shows up in the heightmap edit mode. If you leave it unchecked, it will probe first at the bottom left corner of your grid and treat that as the Z=0 point. 

I am still testing it. (It doesn't issue a G92 for this point, in other words it does not adjust your work coordinates). But if you want to try it out, it is at https://github.com/mjml/Candle in the probe-origin-option branch (forked from Experimental here).