Denvi/Candle #358: G3 code doesn't work

bushse, 2020-04-08:
G3 X141.256 Y3.8 I0. J3.8 F200 < error:33

ondras12345, 2020-04-08:
Hi,
This is a GRBL issue, not Candle issue.

[Error 33](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages):
> The motion command has an invalid target. G2, G3, and G38.2 generates this error, if the arc is impossible to generate or if the probe target is the current position.

For more info about arcs, read this [LinuxCNC documentation page](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3).

MeJasonT, 2020-04-08:
Check cad drawing you might have gaps where your lines and arcs are joined.  Gets me every now and then as well.

MeJasonT, 2020-04-08:
Do you check your Gcode/NC file ?
its good practice and we really should learn how to be proper machinists.
Some times you can spot coordinate errors, they might have massive unexplained gaps or suddnly go negative when you dont expect them to.

so G3 X141.256 Y3.8 I0. J3.8 F200
say the nest move was G3 X138.256 Y3.8 I46. J3.8, if you were expectin a circle the x start coordinate has changed and the arc centre has moved.  It might be even simpler.  

You might be set for inch instead of mm, its woth getting into the habbit of learning your code - just because you can.  It might save you a crash one day.

bushse, 2020-04-08:
[image]
G3 X141.256 Y3.8 I0. J3.8 F200

MeJasonT, 2020-04-08:
zoom as close in as you can to that location and you will probably find the line isnt joined to the arc,   Commands in grbl are buffered ie the sender can send 4 or 5 commands which will be executed in turn.  from your simulation it has obviously done the x axis cut along the bottom and started the corner. i would sugest looking at the node/line that joins the curve on the y axis where your tool is currently located and the next joint/nodes of the next corner.

bushse, 2020-04-08:
> 해당 위치에 최대한 가깝게 확대하면 선이 호에 연결되지 않은 것을 알 수 있습니다. grbl의 명령이 버퍼링됩니다. 즉, 발신자가 4 또는 5 개의 명령을 보낼 수 있습니다. 시뮬레이션에서 바닥을 따라 x 축을 자르고 모서리를 시작했습니다. 공구가 현재 위치한 y 축의 커브와 다음 코너의 다음 조인트 / 노드를 연결하는 노드 / 라인을 보는 것이 좋습니다.

zoom as close in as you can to that location and you will probably find the line isnt joined to the arc, Commands in grbl are buffered ie the sender can send 4 or 5 commands which will be executed in turn. from your simulation it has obviously done the x axis cut along the bottom and started the corner. i would sugest looking at the node/line that joins the curve on the y axis where your tool is currently located and the next joint/nodes of the next corner.

bushse, 2020-04-08:
There is no space between the arc and the straight line by chamfering in the straight corner in the CAD program.

bushse, 2020-04-08:
It is a G code calculated by the cam program in cad drawing

bushse, 2020-04-09:
> 해당 위치에 최대한 가깝게 확대하면 선이 호에 연결되지 않은 것을 알 수 있습니다. grbl의 명령이 버퍼링됩니다. 즉, 발신자가 4 또는 5 개의 명령을 보낼 수 있습니다. 시뮬레이션에서 바닥을 따라 x 축을 자르고 모서리를 시작했습니다. 공구가 현재 위치한 y 축의 커브와 다음 코너의 다음 조인트 / 노드를 연결하는 노드 / 라인을 보는 것이 좋습니다.

Thank you.

bushse, 2020-04-09:
> 안녕하세요,
> 이것은 촛불 문제가 아닌 GRBL 문제입니다.
> 
> [오류 33](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages) :
> 
> > 모션 명령에 잘못된 대상이 있습니다. 아크 생성이 불가능하거나 프로브 타겟이 현재 위치 인 경우 G2, G3 및 G38.2가이 오류를 생성합니다.
> 
> 아크에 대한 자세한 내용은이 [LinuxCNC 설명서 페이지를 참조하십시오](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3) .

Thank you.

bushse, 2020-04-09:
Machine coordinates X0.0 Y0.0 position
G3 X0.0 Y0.0 I0.0 J3.8 F500 Enter
I draw 360 circles mainly on X0.0 Y3.8
I don't know

bushse, 2020-04-09:
Solved
A single command draws a circle
However, when the next command is directed, it finishes at the specified location and executes the next command...
However, there are many commands that are not yet available.
Many of the commands, including the M4, do not understand how to use them.

MeJasonT, 2020-04-09:
your CAM software needs a postprocessor for GRBL, its a file which
translates your cad design into XYZ coordinates for your machine
where grbl does not support a command it gets replaced by lines - so a
circle for example might become a series of lines that made up a circle
not the case as it already covered but basically it creates lines so it can
achieve moves that grbl cant produce due to its limited gcode command set
If your cad file is sending out a default set of commands it may be sending
the commands GRBL does not understand.
do a google search for "your cad programs name"+postprocessor.

bushse, 2020-04-10:
> your CAM software needs a postprocessor for GRBL, its a file which translates your cad design into XYZ coordinates for your machine where grbl does not support a command it gets replaced by lines - so a circle for example might become a series of lines that made up a circle not the case as it already covered but basically it creates lines so it can achieve moves that grbl cant produce due to its limited gcode command set If your cad file is sending out a default set of commands it may be sending the commands GRBL does not understand. do a google search for "your cad programs name"+postprocessor.
> […](#)
> -- *JJ Thompson Engineering & Marine Services* *Westgarth Cottage* *Hayton* *Brampton* *Cumbria* *CA8 9HL* *Telephone 01228 670430*

thank you
I don't know the cam program very well
I don't know if grbl is the first time. 
Now I'm going to try mastercam 2020 version
Information about postprocesor is needed
Can you share it?

bushse, 2020-04-10:
> > your CAM software needs a postprocessor for GRBL, its a file which translates your cad design into XYZ coordinates for your machine where grbl does not support a command it gets replaced by lines - so a circle for example might become a series of lines that made up a circle not the case as it already covered but basically it creates lines so it can achieve moves that grbl cant produce due to its limited gcode command set If your cad file is sending out a default set of commands it may be sending the commands GRBL does not understand. do a google search for "your cad programs name"+postprocessor.
> > […](#)
> > -- _JJ Thompson Engineering & Marine Services_ _Westgarth Cottage_ _Hayton_ _Brampton_ _Cumbria_ _CA8 9HL_ _Telephone 01228 670430_
> 
> thank you
> I don't know the cam program very well
> I don't know if grbl is the first time.
> Now I'm going to try mastercam 2020 version
> Information about postprocesor is needed
> Can you share it?

If you generate an nc code in macam,
X5.575 Y9.575
Z1.
G1 Z-.15 F200.
X8.269 Y9.583 F240.
X8.315 Y9.975
X7.635 Y10.168
X7.951 Y11.784
X8.42 Y12.144
X8.5 Y11.978
X8.64 Y11.633
X8.995 Y10.657
X9.073 Y10.488
X9.159 Y10.325
G3 X9.425 Y9.953 I1.863 J1.051
X9.557 Y9.822 I1.31 J1.182
G1 X9.636 Y9.77
X9.678 Y9.749
X10.167 Y9.617

That's how it's created.
What I want is 
G1 X54.1875
G1 X54.2812
G1 X54.3750 Y12.0000
G1 X54.4688
G1 X54.5625
G1 X54.6562 Y11.9062
G1 X54.7500
G1 X54.8438 Y11.8125
G1 X54.9375
G1 X55.0312 Y11.7188
G1 X55.1250 Y11.6250
G1 X55.2188 Y11.5312
G1 Y11.4375
G1 X55.3125 Y11.3438
G1 Y11.2500
G1 X55.4062 Y11.1562
I want this to happen
It's too much to modify by hand, so it's hard

MeJasonT, 2020-04-10:
GRBL Doesnt do resolution to 4 decimal places ie X 0.000 and not X0.0000
As for post processor
try
https://www.emastercam.com/forums/topic/97978-mastercam-91-w-grbl/
https://en.industryarena.com/forum/grbl-post-processor-mastercam-v8--364940.html
https://mecsoft.com/downloadposts/  Xcarve GRBL in list

that should help
J.

bushse, 2020-04-10:
> GRBL Doesnt do resolution to 4 decimal places ie X 0.000 and not X0.0000 As for post processor try https://www.emastercam.com/forums/topic/97978-mastercam-91-w-grbl/ https://en.industryarena.com/forum/grbl-post-processor-mastercam-v8--364940.html found with search https://www.google.com/search?q=mastercam+grbl+postprocessor&oq=mastercam+grbl+postprocessor&aqs=chrome..69i57.9406j0j7&sourceid=chrome&ie=UTF-8 that should help J.
> […](#)
> -- *JJ Thompson Engineering & Marine Services* *Westgarth Cottage* *Hayton* *Brampton* *Cumbria* *CA8 9HL* *Telephone 01228 670430*

I'm sorry
The question is wrong
It's not a decimal point issue 
You need a way to give each line a g code execution command.

ex G1 X4.123 Y3.2
     X8.12 Y2.34
     G3 X9.425 Y9.953 I1.863 J1.051
     X9.557 Y9.822 I1.31 J1.182 =======> 
    
  To
    
=>G1 X4.123 Y3.2
     G1 X8.12 Y2.34
     G3 X9.425 Y9.953 I1.863 J1.051
     G3 X9.557 Y9.822 I1.31 J1.182

   So

closed 2020-04-10 by bushse