Denvi/Candle #360: Setting Home and CNC not engraving

RLBeadon, 2020-04-09:


RLBeadon, 2020-04-09:
Please can you explain to me how to set the home position? I have pressed the home button and the Z motor drive up without stopping. Is there a way to set limit positions to prevent that happening?

I tried to engrave my name on a piece of wood  when I sent the image to start the motor starts to run but all it does is engrave a square at the start point. 

I see the work coordinates are0 and do not change

[image]
[image]

Settings
$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

MeJasonT, 2020-04-09:
Do you have hardware limit switches on your axis - Best to have them
the following settings are what you are interested in.

$20=0 (soft limits, bool)  you can set the max working area and use this setting to stop your machine running off the end of the world.
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

$21=0 (hard limits, bool)  If you are using hardware limit switches set to =1
$22=1 (homing cycle, bool) This is =1 as you wish to use homing
$23=0 (homing dir invert mask:00000000) If you require your machine to home in a different location you can drive axis in another direction
$24=25.000 (homing feed, mm/min) speed to move to home position
$25=500.000 (homing seek, mm/min) speed to re seek home/switch position
$27=1.000 (homing pull-off, mm) The safe location/clearance away from the homing switch (applies to all 3 axis)

When you send home command (press button) the Z axis moves first it will raise until a limit is reached
which is best set using a limit switch.  Once triggered the Z will lower and will seek the switch again
once found the z will move clear of the switch to your preset safe distance (about 5mm should do it)  $27  dont put it right on the switch or you will alway pick up program halts when the z gets any where near the switch.

X & Y Axis then move to seek home switches.
you must acheive a successful home position or your machine will act irratically as it wont know where it is and how far it can move.

Check the switches are working if you have them installed, they should connect Gnd to the limit switch input. when the switch is made/closed.  switches will be normally open when not triggered.
this is how GRBL is configured. it can be changed so that breaking a switch contact will trigger effectively an alarm $5=0 (limit pins invert, bool).  The limit switches are ignored as alarms when grbl is in homing operation  but once your machine is homed any triggering of these switches triggers an alarm and halts your program.  Once you have your machine successfully homed, jog your axis to where you want your 0,0,0 start coordinates to be on your work piece. and zero the axis I advise not lowering you z axis all the way down to your workpiece until you have confidence in your machine and setup, cut in air above the workpiece, that way you can see its going to work as expected.  Simulate your job (the code/NC file you have created in candle without enabling your machine or use a program like Camotics to simulate your job file.  Do all that you will be well on your way to being a pro.

RLBeadon, 2020-04-10:
@MeJasonT  thank you for your help. I dont have limit switches but want to install some.  is there an instruction video of how to install these and where to connect them on the control board?

ondras12345, 2020-04-10:
Hi, homing is documented on the Grbl wiki. See:
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches
https://github.com/gnea/grbl/wiki/Connecting-Grbl#grbls-pins

RLBeadon, 2020-04-11:
@MeJasonT , @ondras12345 
Can I continue to use the CNC without fitting the limits, I will be fitting the limits just in Lockdown so cant get out

ondras12345, 2020-04-11:
Just disable the homing cycle. $22=0

You don't need homing if you don't need to always have the same machine coordinate system origin after power-up.

RLBeadon, 2020-04-11:
@ondras12345  I have done as instructed and have tried to engrave my name, the bit does not move from  the start position. I did a video and tried to upload the rodney.nc file but am not allowed  as file type is not supported. How else can I show you to get advice

ondras12345, 2020-04-11:
Rename the .nc file to .txt.

What is the machine status? Did you try resetting it after changing the settings? Isn't it locked? https://github.com/Denvi/Candle/wiki/03-States

RLBeadon, 2020-04-11:
@ondras12345 

[Rodney.txt](https://github.com/Denvi/Candle/files/4464343/Rodney.txt)

The status shows running,  The work coordinates all are 0  and dont change

I unlocked and in the console it shows S10000> ok

I closed the programe and reopened it after changing the settings . I am not sure how to reset it 

The machine begins to engrave but does not go past the start point shown by the orange square in the below picture

[image]

ondras12345, 2020-04-11:
> The machine begins to engrave but does not go past the start point ...

Does this mean the machine doesn't move at all? Can you jog it? Does the spindle spin?

Maybe try disabling the feed override for now.

According to your settigns, your machine cannot move faster than 500 mm/min (`$111=500.000 (y max rate, mm/min)`, ...). Your gcode says `G1 F711`, which is too fast. This shouldn't be a problem either (see [$110 - $112](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#110-111-and-112--xyz-max-rate-mmmin)), just don't expect the machine to go this fast.

MeJasonT, 2020-04-11:
@ondras12345 is it possibly that his workcoordinates and machine coordinates are both at 0,0,0 so the machine thinks it cant move?  If his bed was 300x 300 for example would entering -300,-300 in the machine coordinates sort it ?  As if it had been homed.

ondras12345, 2020-04-11:
> @ondras12345 is it possibly that his workcoordinates and machine coordinates are both at 0,0,0 so the machine thinks it cant move? If his bed was 300x 300 for example would entering -300,-300 in the machine coordinates sort it ? As if it had been homed.

$20=0 (soft limits, bool)
Soft limits are disabled. The machine should move regardless of the coordinates.

Maybe he wanted to say that the machine is moving, but the tool **in the simulation** stays at work 0,0,0. That could be caused by using a Candle version newer than 1.0.x with Grbl 0.9.

MeJasonT, 2020-04-11:
Oh that old turkey, id forgotten about the simulation not refreshing issue.

RLBeadon, 2020-04-11:
@ondras12345  and @MeJasonT 

I think I am actually just confusing everything especially myself. 

This is the version of  grblControl that I have 

[image]

To be perfectly honest I dont know where I downloaded the software from, it may have been off the Banggood site where I purchased the 3018 engraving machine from.

Is this the version I must download Windows: candle_1.1.7.zip from  

https://github.com/Denvi/Candle

Is this the only software i must download ?

What has been happening ( I apologise in advance I dont know all the terms)

1.  I can Jog the machine in all directions and the spindial up and down
2. When I send my image for engraving the motor starts to spin. the table moves and the spindle runs and starts to engrave.
3. On the screen the work coordinates are all 0,0,0 and never change.
4. The orange tool image ( I presume) on the screen only works in the box where the red, green and Blue arrows are.

I waited 10 minutes and it did not move from this position.

[Rodney Not engraving.txt](https://github.com/Denvi/Candle/files/4464700/Rodney.Not.engraving.txt)

This is an avi document

I really appreciate all your help I am totally confused

ondras12345, 2020-04-11:
> Is this the version I must download Windows: candle_1.1.7.zip from
> 
> https://github.com/Denvi/Candle
> 
> Is this the only software i must download ?

Yes.

Which Grbl version are you running? It is printed in the console when the machine starts. I _think_ you are running Grbl 0.9, but I'd like to verify.

RLBeadon, 2020-04-11:
> > Is this the version I must download Windows: candle_1.1.7.zip from
> > https://github.com/Denvi/Candle
> > Is this the only software i must download ?
> 
> Yes.
> 
> Which Grbl version are you running? It is printed in the console when the machine starts. I _think_ you are running Grbl 0.9, but I'd like to verify.

its version 0.8.1 I got the details in help where it says about

ondras12345, 2020-04-11:
> its version 0.8.1 I got the details in help where it says about

That's Candle version, not Grbl version. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

RLBeadon, 2020-04-11:
Ok I have just downloaded and installed it its 

Version 1.1.7

ondras12345, 2020-04-11:
That's the version number of the software that is running on your computer (Candle).
I wanted the version number of the **Grbl firmware running on your machine**.
It should be one of the first things in the console textbox in Candle.

RLBeadon, 2020-04-11:
Sorry I have opened and closed it several times the version number of the software does not come up in the console. I dont know where else to look. I am confused between the gblControl and Candel, they both look similar when you open them. I know thats nothing to do with the version

ondras12345, 2020-04-11:
grblControl is an old version of Candle. The project renamed.

Press the 'Reset' button (next to the Unlock button) in Candle an copy the response from the console to your post.

RLBeadon, 2020-04-11:
@ondras12345 

Serial port error 11: The system cannot find the path specified.
[CTRL+X] < Grbl 0.9j ['$' for help]
[CTRL+X] < Grbl 0.9j ['$' for help]
S9600 < ok
[CTRL+X] < Grbl 0.9j ['$' for help]
S9600 < ok

is any of this what you asked for

ondras12345, 2020-04-11:
> @ondras12345
> 
> Serial port error 11: The system cannot find the path specified.
> [CTRL+X] < Grbl 0.9j ['$' for help]
> [CTRL+X] < Grbl 0.9j ['$' for help]
> S9600 < ok
> [CTRL+X] < Grbl 0.9j ['$' for help]
> S9600 < ok
> 
> is any of this what you asked for

Yes, `Grbl 0.9j`.
Your machine should work just fine with Candle 1.0.x https://github.com/Denvi/Candle/releases

Candle 1.1.7 is too new for your machine.

RLBeadon, 2020-04-11:
Do I leave grblControl installed on my machine @ondras12345

ondras12345, 2020-04-11:
grblControl and Candle are both running on your computer only. The CNC machine can work without both, they are just for convenient control of the machine.

Candle 1.0.x is a newer version of grblControl. You can leave both installed, just don't run them at the same time. You shouldn't really need grblControl, though, so you can uninstall it.

Don't use a version of Candle newer than 1.0.x, it won't work with Grbl 0.9.

RLBeadon, 2020-04-11:
@ondras12345 

Hi

So I installed Candel 1.0.10

The firmware is 0.9

I loaded my file for engraving

set my Zero x/y

set my zero set  with the tip of the tool on the surface of the wood.

On the consoule the work coordinates never changed 0,0,0

I pressed Send and the spindle started, it moved to the position and started to engrave.

All that seems to be happening after 5 minutes is that I have what appears to be circle being engraved in the wood.

On the screen I think its called the simulator the orange tool is spinnning but not following the path it shows that it is cutting. but does not follow the cutting path 

$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok
$G < [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
G91G0Y-5.00 < ok
G90 < ok
$G < [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
G91G0Y-5.00 < ok
G90 < ok
$G < [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
G91G0Y-5.00 < ok
G90 < ok
$G < [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
G91G0Y5.00 < ok
G90 < ok
[... 258 lines cut ...]
$# < [G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,-10.774]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
G92X0Y0 < ok
$# < [G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:-110.000,-6.000,-10.774]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok

ondras12345, 2020-04-11:
Try $10=3
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary

RLBeadon, 2020-04-11:
@ondras12345  Thank you ! the tool is following the cut patten and I can see the position on the screen the work coordinates are showing now so I am going to let it run and see that it is all ok.  I really would like to express my sincere gratitude for your assistance and patience.

RLBeadon, 2020-04-11:
@ondras12345 

The machine is  engraving fine but in reverse so the name is coming out backwards, I was trying to engrave my name. It took 4 Hours to engrave the o to 0.6

Is this in the the software used to generate the file for engraving or a machine setting

RLBeadon, 2020-04-12:
@ondras12345  and @MeJasonT  Good day

Below is the image of how my CNC is engraving. It is engraving in the wrong direction like a "Negative" and so very slow. What you see in this picture is 15 hours of engraving. I used   Easeal  to generate this wording.

Can I have some guidance as how to resolve this please 

[image]

ondras12345, 2020-04-12:
Hi, I've got some links for you:
- https://github.com/Denvi/Candle/issues/356
- https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time
- https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask
- http://linuxcnc.org/docs/html/user/user-concepts.html#_machine_configurations

MeJasonT, 2020-04-12:
You still have your axis back to front - you are not alone
https://openbuilds.com/threads/cnc-router-cuts-a-mirror-mage.12683/

MeJasonT, 2020-04-12:
If your not happy - go back to basics and start again.
you have confidence in your hardware and you know it all works.
just add the settings you know that have nothing to do with directions
so bed size, the current accelleration setting you have that work, steps/mm.
then disable homing =0
make jog moves (small values - for safety )
check you are getting them right - only change one at a time and check.
(Just wait till you have to do crap like this on £2 million submersible ROV's).
as @ondras12345 said earlier if your not using homing dont enable it. directions then become irrelevant apparently.

Settings and sample values | Description
-- | --
$0=10 | Step pulse, microseconds
$1=25 | Step idle delay, milliseconds
$2=0 | Step port invert, mask
$3=0 | Direction port invert, mask
$4=0 | Step enable invert, boolean
$5=0 | Limit pins invert, boolean
$6=0 | Probe pin invert, boolean
$10=1 | Status report, mask
$11=0.010 | Junction deviation, mm
$12=0.002 | Arc tolerance, mm
$13=0 | Report inches, boolean
$20=0 | Soft limits, boolean
$21=0 | Hard limits, boolean
$22=1 | Homing cycle, boolean
$23=0 | Homing dir invert, mask
$24=25.000 | Homing feed, mm/min
$25=500.000 | Homing seek, mm/min
$26=250 | Homing debounce, milliseconds
$27=1.000 | Homing pull-off, mm
$30=1000. | Max spindle speed, RPM
$31=0. | Min spindle speed, RPM
$32=0 | Laser mode, boolean
$100=250.000 | X steps/mm
$101=250.000 | Y steps/mm
$102=250.000 | Z steps/mm
$110=500.000 | X Max rate, mm/min
$111=500.000 | Y Max rate, mm/min
$112=500.000 | Z Max rate, mm/min
$120=10.000 | X Acceleration, mm/sec^2
$121=10.000 | Y Acceleration, mm/sec^2
$122=10.000 | Z Acceleration, mm/sec^2
$130=200.000 | X Max travel, mm
$131=200.000 | Y Max travel, mm
$132=200.000 | Z Max travel, mm

MeJasonT, 2020-04-12:
The defaults are not usually a million miles away to get most users started
What are you doing thats different ?  where do you think your machine should be located at the start of a job 0,0 ref and what direction do you expect it to work in from that position.  the phsyical world need to be sorted in your head and withn the mechanics of your machine before you attempt to set its jog directions.
Stepper motors can be wired back to front which will gaurantees it runs backwards on you,  with $3 command these little challenges can be sorted.

ondras12345, 2020-04-12:
The table MeJasonT sent you is for Grbl 1.1. I _think_ the configuration for axis inversion is the same as in Grbl 0.9, but you should better check out this link for other settings: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

ondras12345, 2020-04-12:
For the 15hour engraving time, post your Easel config and the tool you are using. It should be possible to optimize it.
Based on how many moves your gcode uses to engrave the letters, I think you are either using a tool that is too small for this job and/or you've set your tool size incorrectly in Easel.

RLBeadon, 2020-04-13:
Hi @ondras12345  and @MeJasonT 

I have the two issues which you are assisting me with. So that I dont confuse myself I am going to answer them in seperate posts.

Firstly dealing with the settings in Easel 

Bit 30 Degree 

[image]

Feed Rate 711.2 mm/min

Plunge Rate 228.6 mm/min

Depth per pass 0.2mm

Fill method 0.2mm  ( Pro Y - axis)

Machine
X-Carve
Work Area  X -284mm 
                   Y -284 mm

Spindel control Manual

Machine details I have not set  so I have set them now

Machine Type , x-Carve other options  shapeoko 1or 2 or other grbl
Motion controller Auduino & gShield
Rail Size  I dont know what this is refering to 
Lead Screw  options ACME thread rod or the other discontinued in 2016 so chose ACME thread rod
Spinddle 24 Volt
Have not checked Dust Shoe

Material I am using Pine so have set it to Sft Mapel 

I have changed the file extension to txt.

[Rodney.txt](https://github.com/Denvi/Candle/files/4469126/Rodney.txt)

RLBeadon, 2020-04-14:
@MeJasonT @ondras12345 

I have gone through the settings I have checked absoluetly everything.  I inverted the x axis  and it went right instead of left and left instead of right  but then it didnt print mirrored it printed correctly.

Checked the settings got the x axis traveling in the correct direction  but when engraving it was mirrored, I am totally out of ideas now. 

I think the only option now is to change the connections on the Hardwiring of the stepper motor.  This is where I am confused alot.

Do I just change the wires +Ve and -Ve  or the other 2 wires as well 

Really confused and frustrated

ondras12345, 2020-04-15:
[image]
These arrows should match the **tool** movement directions.
[image]
Once you make your axes move in the right directions, the text should not be mirrored. It should match what you see in the simulation preview in Candle.
Swapping the wires on the stepper motor connectors is exactly the same as changing $3.

For the engraving time, I think this tool is way too small for this task and your 'depth per pass' could maybe be a bit higher for wood.

RLBeadon, 2020-04-15:
I have changed the wires around

Orginal
B                                         A                               2B                                      2A
Black                                Green                         Red                                     Blue

After swop
B                                         A                               2B                                      2A
Red                                  Blue                             Black                              Green

Then by Jogging and using the table to invert the directions I got all the directions correct.

Sent off the image for engraving and no diffrence 

@MeJasonT , @ondras12345

RLBeadon, 2020-04-15:
> [image]
> These arrows should match the **tool** movement directions.
> [image]
> Once you make your axes move in the right directions, the text should not be mirrored. It should match what you see in the [...]

I think I see my mistake now, I will rectify it and confirm

RLBeadon, 2020-04-15:
@ondras12345 

The frustration continues I see in the past my errors and have tried to rectify them 

I have jogged the motor to confirm the direction

to start X is correct direction, Y is correct direction , Z is the wrong direction, so I entre $3=4

I then jog to check direction again

X is wrong direction, Y is wrong direction Y is correct direction to rectify this I entre $3=4
and it goes back to   X is correct direction, Y is correct direction , Z is the wrong direction

So its just a circle round and round.

I thought I should restart the software, I did same result, I thought I should re boot the computer and re start the software, same result.

I am totally confused now

ondras12345, 2020-04-15:
I don't understand your last post.
> X is wrong direction, Y is wrong direction Y is correct direction

I guess you wanted to say 'Z is correct direction'.
You also wrote $3=4 twice. It wouldn't change anything the second time you ran it if that was the case.

Set $3=0, determine which axes need to be inverted and use the [lookup table](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#2--step-port-invert-maskbinary) to determine what you need to set $3 to.

RLBeadon, 2020-04-15:
@ondras12345  Thank you very much, the machine is jogging in the right direction and engraving correctly.

I will close this shortly.

Before I do this can you direct me where I learn how to set the start point of the engraving and to see where the engraving will be I dont know where my start point is and I dont know where the position of the image is

ondras12345, 2020-04-15:
You can use the G92 offsets to set your work coordinates origin. Just press the 'Zero XY' and 'Zero Z' buttons in Candle.

You can see the position of your work relative to the origin in the simulation in Candle. The origin (0,0,0 point) is where the red, green and blue lines (axes) intersect.
In the bottom left corner, you can see the min and max coordinates for each axis found in your gcode. Your work is within this range, the tool won't go past it.

RLBeadon, 2020-04-15:
> You can use the G92 offsets to set your work coordinates origin. Just press the 'Zero XY' and 'Zero Z' buttons in Candle.
>
> You can see the position of your work relative to the origin in the [...]

Thank you very much

closed 2020-04-15 by RLBeadon