Denvi/Candle #362: Scaling, Is there a setting

RLBeadon, 2020-04-16:
[Name.txt](https://github.com/Denvi/Candle/files/4489431/Name.txt)

I have been trying to engrave the above ( Changed the extension to .txt from .nc)

When I drew the image in Easel the size was 47mm x 22mm .
I then Generate the gCode in easel and export it, opening it in  Candle 1.0.10

When sending it for engraving the actual size it engraves is  28mm x 9mm 

I am trying to find where the size change is, between Easel and Candle

Thanking you in Advance

svenhb, 2020-04-17:
If I import your textfile, I get a dimension of 60 mm x 30 mm
[image]

RLBeadon, 2020-04-17:
@svenhb 

Thank you for replying

I suspect that my settings in step/mm are wrong, 

They currently at 800.000 steps per mm  

This information I have no idea if its true because I have been reading and just think 

I was going to do Trial and erroe adjustments but would really like  know the facts.

What software is this picture above taken from

Best wishes

Rodney

svenhb, 2020-04-17:
Draw a 10mm x 10mm square and measure what you get. Then scale your step / mm accordingly.
The code:
```F1000 (Setup - GCode-Header)
G00 Z2.000
M3 S20000 
G04 P3 
(<Figure Id="1" >)
G00 X0.000 Y0.000  
G01 Z-2.000 F400 
G01 X0.000 Y10.000 F800 (cw 1)
G01 X10.000 Y10.000  (cw 2)
G01 X10.000 Y0.000  (cw 3)
G01 X0.000 Y0.000  (cw 4)
G00 Z2.000 
(</Figure>)
M05
G0X0Y0 (Setup - GCode-Footer)
M30```

svenhb, 2020-04-17:
Probably the right setting is 480 steps/mm?
28 mm * 800 / 47 mm

RLBeadon, 2020-04-17:
> Draw a 10mm x 10mm square and measure what you get. Then scale your step / mm accordingly.
> The code:
> 
> ```
> G00 Z2.000
> M3 S20000 
> G04 P3 
> (<Figure Id="1" >)
> G00 X0.000 Y0.000  
> G01 Z-2.000 F400 
> G01 X0.000 Y10.000 F800 (cw 1)
> G01 X10.000 Y10.000  (cw 2)
> G01 X10.000 Y0.000  (cw 3)
> G01 X0.000 Y0.000  (cw 4)
> G00 Z2.000 
> (</Figure>)
> M05
> G0X0Y0 (Setup - GCode-Footer)
> M30```
> ```

Hi How do I view this code

svenhb, 2020-04-17:
I don't know - so I never used Candle...
Just copy this GCode to a file and load?

RLBeadon, 2020-04-17:
> I don't know - so I never used Candle...
> Just copy this GCode to a file and load?

Thanks, I have drawn the square 100mm x 100 mm, when engraving it came out 40mm x 40 mm

svenhb, 2020-04-17:
Ok, this means you need more (100/40) steps to reach the 100 mm.
Try to set $100 = 2000 and draw this square again