Denvi/Candle #364: CNC runs then stops strange noise then runs again after changing settings

RLBeadon, 2020-04-20:
Good Day

I have changed the settings 

$100= 2000.000
$101= 2000.000
$102= 2000.000

I changed this as he cnc was not cutting the correct length.

Since then it starts to move, then stops makes a strange sound and then moves again.

I have a video but dont know how to upload it

RLBeadon, 2020-04-20:
Here is a drop box link

https://www.dropbox.com/s/ghn2v248ftiq25x/CNC%20vid.mp4?dl=0

svenhb, 2020-04-21:
I assume you exceed the max. possible speed for your setup.
Did you move manually or somehow controlled?
Which movement command was used? G0 or G1?
What's your settings for $110, $111, $112? Perhaps they are too high.

RLBeadon, 2020-04-21:
Hi @svenhb 

I can Jog the machine using the 10 steps setting on the X and Y axis across the whole machine, If I try to use the 100 steps this is when the problem starts.

"Which movement command was used? G0 or G1?" I am not sure what you mean by this I use the **_Jog_**  on the program to move.

$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)

My machine is a CNC engraving machine 3018 I am using Candle Version  1.0.10

I started changing these settings because  my engraving was scaled down, I drew a square 100mm x 100mm when it engraved it was 40mm x 40mm  so a scale of 10:4. the settings were 800 orginally, I increased it to 2000 these are the settings 

$100= 2000.000
$101= 2000.000
$102= 2000.000

This is when the problem started

RLBeadon, 2020-04-21:
I have done a test square 100 mm x 100mm and it has worked perfectly after changing all the settings back to 800.000 not sure why but its working

svenhb, 2020-04-21:
Now it also draws 100 x 100?
Strange