Denvi/Candle #367: v1.2 French translation

adnovea, 2020-04-25:
[Candle_fr.ts.txt](https://github.com/Denvi/Candle/files/4532662/Candle_fr.ts.txt)

For some reason, the string "Continuously" for the joypad is not translated despite present in the file.

adnovea, 2020-04-25:
[LICENSE.txt](https://github.com/Denvi/Candle/files/4532711/LICENSE.txt)

Here is a modified file to replace the LICENSE in the candle root folder in order to have the Error & Alarm codes as well as the GPL license when clicking on Help.
(_remove the txt extension before copying_)

ondras12345, 2020-04-25:
> [LICENSE.txt](https://github.com/Denvi/Candle/files/4532711/LICENSE.txt)
> 
> Here is a modified file to replace the LICENSE in the candle root folder in order to have the Error & Alarm codes as well as the GPL license when clicking on Help.
> (_remove the txt extension before copying_)

The LICENSE file should only contain the license.
Error and alarm codes depend on the version of Grbl. The LICENSE file should not be modified each time a new Grbl version is released.
It's certainly a good idea to include this information in the software, but it shouldn't be in the LICENSE file.