Denvi/Candle #369: Advices for Candle configuration with Stepcraft CNC

adnovea, 2020-04-25:
[image]

GRBL / Candle configuration for Stepcraft CNC is a nightmare.
Despite the numerous parameters in GRBL, the configuration is not that easy.

I have set the following parameters :

> $0=25  (step pulse, usec)
> $1=50 	(step idle delay, msec)
> $2=0 	(step port invert mask:00000000)
> $3=0 	(dir port invert mask:00000110)
> $4=0 	(step enable invert, bool)
> $5=1 	(limit pins invert, bool)                // Active high
> $6=0 	(probe pin invert, bool)
> $10=1 	(status report mask:00000011)  // need MPos for Candle
> $11=0.010 	(junction deviation, mm)
> $12=0.030 	(arc tolerance, mm)
> $13=0 	(report inches, bool)
> $20=1 	(soft limits, bool)                        // Enable soft limits for security
> $21=0 	(hard limits, bool)
> $22=1 	(homing cycle, bool)                  // Enable Homing cyle
> $23=1 	(homing dir invert mask:00000001)
> $24=500.000 	(homing feed, mm/min)
> $25=2000.000 	(homing seek, mm/min)
> $26=250 	(homing debounce, msec)
> $27=1.500 	(homing pull-off, mm)      // Retract 1.5 mm
> $30=25000                                          // Kress FM1050FME
> $31=5000
> $32=0
> $100=133.333 	(x, step/mm)
> $101=133.333 	(y, step/mm)
> $102=133.333 	(z, step/mm)
> $110=3000.000 	(x max rate, mm/min)
> $111=3000.000 	(y max rate, mm/min)
> $112=1000.000 	(z max rate, mm/min)
> $120=100.000 	(x accel, mm/sec^2)
> $121=100.000 	(y accel, mm/sec^2)
> $122=100.000 	(z accel, mm/sec^2)
> $130=417.000 	(x max travel, mm)  // SC2/600
> $131=597.000 	(y max travel, mm)
> $132=137.00       (z max travel, mm)

But I was not able to match the Stepcraft Machine Space as shown on the pics above.

To overcome such issue, I have "forced" the **Working Space** using the following G10 and stored the values in the EEPROM.

> G10 L2 P1 X-415.5Y-598.5Z-138.5

This makes the **Working space** matching the **Stepcraft space** but not its Machine Space.
Sure, this does not prevent the CNC from working but it's confusing.

Does someone have advice for improving the situation ?
Many thanks

ondras12345, 2020-04-25:
Hi,
I believe that the positive directions of the axes shown by the arrows in picture 2 are wrong. See [Grbl wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time) and [a picture in LinuxCNC docs](http://linuxcnc.org/docs/html/user/user-concepts.html#_machine_configurations).

I don't think it is possible to configure Grbl to behave according to picture 1. You don't really need to care about machine coordinates, though.

Which Grbl version are you using?

adnovea, 2020-04-25:
Hi,
You are correct about the picture. I fixed it.
Thanks for the links. They are helpful.

My GBRL version is 1.1h