Denvi/Candle #370: Candle will only work from CD

Topp53, 2020-04-25:
I have just bought a CNC 3018 engraving machine and it has Candle on the CD. Candle works fine when the cd is in the laptop but I cannot load it from the cd to the laptop. 
   I have downloaded Candle from github Onto my laptop and everything seems to work fine but even though it says it’s connected no actions happen?

ondras12345, 2020-04-25:
Which version of Grbl is your machine running?
Which version of Candle did you download? What was on the CD?

Topp53, 2020-04-26:
Hi, the information on the disk is GRBL 1.1 and Candle 1.1.17. The software on the disk hasn’t load and I haven’t found an .exe file on it anywhere so I may not have GRBL installed on the laptop. 
  I will download it and install.  
Thanks for the reply

ondras12345, 2020-04-26:
GRBL is the firmware that runs on your machine.

Topp53, 2020-04-26:
I have downloaded Candle 1.1.17. 
When I use Candle from my laptop it connects ok to com 8 and it’s shown connected in the upper right corner. I can start and stop the spindle rotating but if I try to move along any axis xyz I just get a buzzing. Also when I try to jog it along I get xyz locations indicated

closed 2020-04-26 by Topp53

ondras12345, 2020-04-26:
> ... but if I try to move along any axis xyz I just get a buzzing

This is most likely an issue with your Grbl settings. See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Topp53, 2020-04-26:
Thanks I will have a look, thanks for taking the time to answer

DarrenSchwartz, 2020-04-27:
Actually this sounds like a similar problem I had with my cheap router from amazon, but only on one axis. It turned out that the manufacturer wired the motor connectors wrong. Check each of the 4-wire harnesses between the motor and control board to make sure none of them are crossed. It happened to me.

Topp53, 2020-04-27:
Thanks for that Darren, something else to check

Topp53, 2020-04-27:
Well I’ve tried all suggestions but nothing works, though I have found the steppers are actually moving but very slowly.