Denvi/Candle #371: 3018 cnc open port issue

Rabid-terd, 2020-04-27:
Hi i am new to this stuff and need some help and have no where else to go, i set up my cnc and installed the drivers, gbrl laser, and candle programs that came with it. But when i try and use it and select the port its on it comes up open port and not idle. is there something i should look for that im missing?

many thanks

ondras12345, 2020-04-27:
Your CNC may be missing the [Grbl firmware](https://github.com/gnea/grbl/wiki/Compiling-Grbl). Try connecting with a terminal program such as [PuTTY](https://www.putty.org/). Refer to https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#getting-started for the serial port config and basic commands. If the machine does not respond, it's most likely a problem with the hardware or missing firmware.
Post a foto of the PCB that is on the back of the machine.
If it is the Woodpecker control board, it should have a few LEDs on it. The LEDs labeled 24V, 12V, and 5V should all be lit. The ones labeled RxD and TxD should blink during the serial communication.

Rabid-terd, 2020-04-27:
I have no blinking lights and it says serial port error 1 no file or directory it wont take $ command. when unplugged from laptop I have T R P lights on and when I plug in only P light is on. I tried multiple versions of candle for this with same result?

ondras12345, 2020-04-27:
Did you select the correct serial port in Candle settings?

Rabid-terd, 2020-04-27:
yes it just says open port wont let me do anything

ondras12345, 2020-04-27:
> serial port error 1 no file or directory

This is printed when Candle can't open the specified serial port.

Rabid-terd, 2020-04-27:
I can find the machine in device management options like the cpu is seeing the machine and also installed recent driver, do you think it is just candle and I should try a different grbl?

ondras12345, 2020-04-27:
https://doc.qt.io/qt-5/qserialport.html#SerialPortError-enum
Error 1 means "An error occurred while attempting to open an non-existing device.".

Try opening the serial port with PuTTY, Hyperterminal, or a similar program.

Rabid-terd, 2020-04-27:
I have putty up but have no idea how to use it lol

 
 

https://doc.qt.io/qt-5/qserialport.html#SerialPortError-enum

Rabid-terd, 2020-04-27:
https://doc.qt.io/qt-5/qserialport.html#SerialPortError-enum

ondras12345, 2020-04-27:
Try using another USB cable. If it doesn't help, restart the computer. The serial port may be held open by another process.

Rabid-terd, 2020-04-27:
alright I opened it and its just a blank black screen?

ondras12345, 2020-04-27:
So PuTTY didn't display any errors this time? Did you configure the serial port correctly (baud rate, disable flow control)? Grbl should print the version when you open the serial port.

Rabid-terd, 2020-04-27:
it reads no errors on putty but candle just says disconnected now wont even go to port opened now

ondras12345, 2020-04-27:
You have to terminate any other processes that are using the serial port (such as putty) before you try opening Candle. Does Candle still display the same error message? Are you absolutely sure you selected the correct COM port in the drop-down list in Candle settings?

Your issue is not related to Candle if the machine doesn't respond when you connect to it using PuTTY. When you connect and start typing random text (it won't appear on the screen), do any LEDs on the machine blink? Please post a photo of the PCB.

Rabid-terd, 2020-04-27:
On Monday, April 27, 2020, 09:39:16 AM EDT, keith sager <moparstar18@yahoo.com> wrote:

Rabid-terd, 2020-04-28:
I have been getting this far, they sent me new firmware and to use this x loader but this is as far as it will go. don't know why this is such a pain in the ass

    On Monday, April 27, 2020, 09:41:09 AM EDT, keith sager <moparstar18@yahoo.com> wrote:  
 
  

    On Monday, April 27, 2020, 09:39:16 AM EDT, keith sager <moparstar18@yahoo.com> wrote:

Rabid-terd, 2020-04-28:
have serial port error but says im connected now... am I close to something here lol

    On Tuesday, April 28, 2020, 08:28:52 AM EDT, keith sager <moparstar18@yahoo.com> wrote:  
 
  
I have been getting this far, they sent me new firmware and to use this x loader but this is as far as it will go. don't know why this is such a pain in the ass

    On Monday, April 27, 2020, 09:41:09 AM EDT, keith sager <moparstar18@yahoo.com> wrote:  
 
  

    On Monday, April 27, 2020, 09:39:16 AM EDT, keith sager <moparstar18@yahoo.com> wrote:

rcchristensenjr, 2020-11-20:
It took me a couple of days but finally decided to disconnect manual controler. It ended up it was interfering with usb connection. After disconnecting data line worked and could install what i needed!!!!!!

Glasairman, 2021-07-13:
What worked for me is that I started Candle as "Administrator" . The Port changed from Open to Idle and JOG worked at long last. I got the idea from a post on the AskUBUNTU forum where the solution was to start Candle as ROOT. And as Admiinistrator  is the Windows equivalent of ROOT, I gave it a try..

Glasairman, 2021-07-13:
Looks like Windows is introducing more and more of the security aspects of Linux device ownership at user level. Influence of RedHat maybe?

SullivanDesign, 2021-08-29:
I seem to be in a similar situation as OP.  New to all this and have the same open port when starting candle and connecting to the correct com port.

Trying to connect with candle running on windows 10;
first checked device manager port says connected to device and is working properly
restarted computer and rechecked device manager
connected via PuTTY; it seems to be connecting at least somewhat? but it's just a blank screen can't type anything
[image]
I've double and triple checked I'm connecting to right port on candle
The only led's lighting up on board is USB and 24v
[image]

Not sure what to try next. Anyone have ideas?

pasqubp, 2021-09-11:
Sorry, I don't know English, so I used a translator, I had the same problem, the solution is to disconnect the screen than bring the cnc to use it manually

boogy886, 2022-10-09:
> I seem to be in a similar situation as OP. New to all this and have the same open port when starting candle and connecting to the correct com port.
> 
> Trying to connect with candle running on windows 10; first checked device manager port says connected to device and is working properly restarted computer and rechecked device manager connected via PuTTY; it seems to be connecting at least somewhat? but it's just a blank screen can't type anything I've double and triple checked I'm connecting to right port on candle The only led's lighting up on board is USB and 24v [image] Not sure what to try next. Anyone have ideas?
> 
> Edit: Bitrate defaulted to 9600 for the port while it was set to 115200 in candle. Once I increased it through device manager they started communicating properly... I think.
> 
> Edit2: It went from open port to idle, and console said it unlocked when I pressed the button, but then nothing responded. Restarted the program and it went back to "port open" rechecked baud rate was identical. Guess looking for the next problem.
> 
> Edit3: Seems to go through "port open" --> "alarm" --> "idle" only if the usb is connected but power to the cnc is turned off (only power to the board through USB). As soon as I turn the power on the idle goes back to "port open"

I am having the exact issue as Edit3...did you ever resolve this?