Denvi/Candle #375: Linux issues

mojuave, 2020-05-06:
I have candle running on linux mint. I have a few issues I can not see the visualizer its just a blank screen with cursor moving, 2 it stops on me was almost like resetting the port so I removed a usb drive and seemed to work but then just stopped spindle stayed running it just stopped moving. I am trying a code but just milling air atm and its making odd noises(hooked to windows comp it didnt do it) almost like some grinding in the motor possibly ramping up and down? 

Can I dedicate a port in linux, or just always make sure I have nothing else plugged in?

Is there a better linux to run this on?

mojuave, 2020-05-06:
Seemed to be going good so started a carve and 22 minutes in went to "serial port error 9: resources temporarily unavailable" what am I missing

DarrenSchwartz, 2020-05-06:
I had a similar issue when getting started with my new machine and found some advice that fixed it for me. Turns out the motors throw a lot of electromagnetic interference and it would work fine for a few minutes and then hang with the spindle on. 

I did a couple of things and the issue has not recurred although I have not experimented to find out which of the fixes are really needed. First, I covered the controller board with a foil lined box to protect the microcontroller. Second, I used the best quality USB cable I could find which had ferrite cores on it and seemed to have good shielding. Some USB cables are really crap and have little or no shielding. Third, I wrapped most of the motor wire bundles in foil and separately wrapped the USB cable in foil. 

I'm not sure all the motor wire foil is needed. I think focusing on the USB cable and shielding the control board were the best bang. But it was night and day for me. Went from not getting through a single job to never having a problem.

mojuave, 2020-05-06:
If I run using a windows laptop that doesn't happen. Using windows I have to dedicate the carve and cant use laptop, my processor not that good to multitask. So had an old computer put Linux on it I believe it has something to do with that

ondras12345, 2020-05-06:
What power brick are you using to power the CNC machine? Try plugging it to the same mains socket as the computer. I think this may be a ground loop problem.