Denvi/Candle #380: User Commands Not Working inside Candle 1.2.8b

vcholman, 2020-05-11:
Hello, 

This is driving me crazy, I can't get any of the user commands 1-4 inside the settings menu to work. 
I am trying to enter 2  short  GCode scripts assigned to Buttons 1 and 2. 

The scripts are for finding Z height and XY zero position.  The software keeps stopping and displays ALARM4-4 after trying to execute a G38. 2 command. 

Here are the GCode commands I want  to assign to user command button #1 
G21
G38.2 Z-20 F50
G92 Z-13.75
G91
G0 Z10

This is the alarm message I get from the console

G21G38.2 Z-20 F50G92 Z-13.75G91G0 Z10 < error:24

There is no movement, just the Alarm.

Hope someone  understands this and can help.

ondras12345, 2020-05-11:
Hi,
See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages
```
error:24 Two G-code commands that both require the use of the XYZ axis words were detected in the block.
```

You should be able to use `;` as a command separator https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3841

vcholman, 2020-05-11:
I not a programmer, so I need additional assistance in understanding your reply,

Would you please respond with an example as what I need to type to execute the following commands on a single line. 

G21
G38.2 Z-30 F50
G92 Z 13.75
G91
G0 Z10

Thanks,

ondras12345, 2020-05-11:
G21;G38.2 Z-30 F50;G92 Z 13.75;G91;G0 Z10

vcholman, 2020-05-12:
Hi Ondras12345,

I tried your example on the Probe Commands line inside the settings menu. Still not working. Below is a copy of the alarm message returned.

The attached jpg files shows a capture of my settings menu

[image]

G21 < ok
G38.2 Z-30 F50 < ALARM:4
ok
G92 Z 13.75 < error:9
G0 Z10 < error:9
$# < [G54:-160.000,-80.000,-10.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok

This should not be this difficult to solve.

Hope you can help

ondras12345, 2020-05-12:
```
ALARM:4 Probe fail. The probe is not in the expected initial state before starting probe cycle, where G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.
```

```
error:9 G-code locked out during alarm or jog state
```

http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38

vcholman, 2020-05-13:
Problem solved.
Apparently GRBL 1.1 in my EEPROM got corrupted. I used XLOADER to reinstall.
G38.2 and User Commands are now working.