Denvi/Candle #381: nothing works - CNC3018

phil123456, 2020-05-11:
Hello,

I try to connect to the cnc, but nothing seem to work, I click here and there, I see commands beeing written in the console
but the CNC wont move at all

also how can I set it to english ? cos half of the french translations are missing and I prefer english anyway

sheers

ondras12345, 2020-05-12:
Which Grbl and Candle version are you using?

Braunfeltd, 2020-05-12:
1.2b1

ondras12345, 2020-05-12:
What Grbl version? Grbl 0.9 is not compatible with this version of Candle.

After connecting to your machine, a text like this should appear in the console (in Candle): `Grbl 1.1f ['$' for help]`.

ondras12345, 2020-05-12:
> * also how can I set it to english ? cos half of the french translations are missing and I prefer english anyway

See https://github.com/Denvi/Candle/issues/54

phil123456, 2020-05-16:
> 
> 
> What Grbl version? Grbl 0.9 is not compatible with this version of Candle.
> 
> After connecting to your machine, a text like this should appear in the console (in Candle): `Grbl 1.1f ['$' for help]`.

what version of grbl should I use ?
I just broke the usb connector of the controller board, so if anyone knows a compatible board for a chinese CNC3018 that would work awesome with Candle, I am in

thanks

phil123456, 2020-05-16:
> 
> 
> > * also how can I set it to english ? cos half of the french translations are missing and I prefer english anyway
> 
> See #54

thanks, I'll do so

123swk123, 2020-05-27:
> I just broke the usb connector of the controller board, so if anyone knows a compatible board for a chinese CNC3018 that would work awesome with Candle, I am in

these boards(GRBL 1.1J) works with cnc3018

https://www.aliexpress.com/item/4000163253874.html?spm=a2g0o.productlist.0.0.1b325ce6mL4JWu&algo_pvid=a856b98a-e25a-4826-b763-4ec9ad2ebf88&algo_expid=a856b98a-e25a-4826-b763-4ec9ad2ebf88-2&btsid=0be3764515905583781008675ee860&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

https://www.amazon.com/MYSWEETY-Control-Engraving-Controller-Engraver/dp/B07S2R59N6/ref=sr_1_3?dchild=1&keywords=grbl+1.1j&qid=1590558553&sr=8-3

you just have to configure the settings before you start using it with the CNC. basically you have to modify the steps/mm.

phil123456, 2020-07-05:
thanks a lot dude
do you mean mod the settings on the board or in the software ?

phil123456, 2020-07-24:
I installed the new bord, it works with bCNC but with Candle, the axes dont move

[image]

ByteHamster, 2021-01-10:
Are you sure that you selected a feed rate from the list below the arrows? I constantly keep forgetting that and then wonder why nothing happens.