Denvi/Candle #384: PROGRAM DONT START, FIXED on old version, but new ver, dont work also..

OZ2CPU-thomas, 2020-05-15:
i used version 1.0.11 is what it repport in about for several weeks
BUT suddenly one day, it just refuse to start, nothing happens..
after a long time of frustrations, i deleted the settings file, and now it starts and works again..
very odd..
also tried to download a new version, but the latest 128B refuse to run, windows version.
it say  : there is no disc in the drive 
please insert a disc into drive
\device\hardisk1\dr1.

I am running windows 7 home

ondras12345, 2020-05-15:
See https://github.com/Denvi/Candle/issues/173#issuecomment-335488688
I don't know if it still applies, though.

OZ2CPU-thomas, 2020-05-15:
thanks mate, deleting the settings file, will let the program start, what i found out was that next time i close it, it refuses to start again, and i must delete the settings file again.. that was odd. happened from one day to another, i did not change behaviour in using it.. so to solve this, i simply deleted the folder "translations" and now it starts again and again, and i dont need to delete my settings file anymore.. so there is a little bit unstable bugs in this program.

sv7qi, 2022-03-17:
Hello, i try to connect my 3018 CNC with candle and Ugs on AMD 64 bytes HP laptop DESKTOP-CJ5IH41, AMD Ryzen 5 2500U with Radeon Vega Mobile Gfx     2.00 GHz, 8,00 GB RAM, the Programme start ok, but i cand load files freezing monitor on both 2 programs. Th programs putting in file C:\Program Files (x86).... any help please?