Denvi/Candle #385: FEATURE SUGGEST, accept low resolution screens

OZ2CPU-thomas, 2020-05-15:
many people use outdated pc hardware to handle their machines
and so do i, this type of old equipment we get for free, and it dont matter if it is broken
by saw dust or metal mill parts.
however old PC hardware often use smaller resolution screens like 768 pixels in the total height !!
this program is close to impossible to use at this low resolution..
how about make a low res screen enable feature ?
so unused features like spindle, heightmap, user commands can go away for real
so scrolling up and down is not needed ? the JOG and feed rate is important..

DarrenSchwartz, 2020-05-15:
I would agree that low resolution support would be useful for the same reason. I use old laptops for this and there is a lot of scrolling involved. Maybe even just reducing some padding and the size of some UI widgets would be enough. Or make those variables user configurable.

izanette, 2020-06-03:
Did you try the beta version? It is much better in this regard.

DarrenSchwartz, 2020-06-11:
Not yet.  Are binaries available somewhere or do I have to build?

ondras12345, 2020-06-11:
https://github.com/Denvi/Candle/releases
Assets of the pre-release