Denvi/Candle #388: machine coordinates!!!

gabrail-110, 2020-05-21:
The machine coordinates do not exceed 999,000 as you can see in the picture
[image]

ASHIT-AXAR, 2020-05-25:
true. also it's working correctly, but doesn't show in the coordinate display.

Denvi, 2020-06-07:
Fixed in [1.2.9b](https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2.9b.zip).

gabrail-110, 2020-06-08:
Hi, thank you, but there has been no change in the software
[image]
[image]

Denvi, 2020-06-09:
Ok, how about [1.2.9.1b](https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2.9.1b.zip)

gabrail-110, 2020-06-09:
It works properly

gabrail-110, 2020-06-09:
The software works great in terms of communication with the microcontroller and displays visual movement, since I'm just a user and I'm not a programmer, but I'm interested in learning programming.