Denvi/Candle #389: Desperately needs "Go to 0,0,0" button

ngreenou, 2020-05-24:
V1.1.7: The "home" button is useless for most of us without limit switches on our cheapo machines. Please add a configuration option for the "home" button to allow us to assign something like "G00 Z10; G00 X0Y0" that will restore the machine to work coordinates 0,0,<something>. 
Thanks!

Braunfeltd, 2020-05-24:
that is what macros are for :)

ASHIT-AXAR, 2020-05-25:
"G0Z10" is right there, called "Safe Position" command. you just need to edit the command at your own will.

[image]

the second one, "G0X0Y0". we hope this highly useful command to be added to "Control" section buttons.
for now, you can temporary assign "G0X0Y0" to one of the User Buttons.

ngreenou, 2020-05-25:
Thanks for the hint!
Nevell

Braunfeltd, 2020-05-25:
sweet learned something new myself from this :)